rizacelik/STM32F411CEU6_INAV_Firmware #1: no gyro or sensors

gazza2577, 2021-11-22:
# status
System Uptime: 80 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 8.68V (3S battery - CRITICAL)
CPU Clock=96MHz
STM32 system clocks:
  SYSCLK = 96 MHz
  HCLK   = 96 MHz
  PCLK1  = 48 MHz
  PCLK2  = 96 MHz
Sensor status: GYRO=UNAVAILABLE, ACC=UNAVAILABLE, MAG=UNAVAILABLE, BARO=UNAVAILABLE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE, IMU2=NONE
Stack size: 6144, Stack address: 0x20020000, Heap available: 2048
I2C Errors: 0, config size: 7049, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = none, used = none
   CURRENT : configured = ADC 1, used = none
  AIRSPEED : configured = none, used = none
System load: 13, cycle time: 1009, PID rate: 991, RX rate: 49, System rate: 10
Arming disabled flags: ANGLE CAL HWFAIL RX CLI
VTX: not detected

rizacelik, 2021-11-23:
Hello gazza2577.

Before connecting the GY-87 Module, connect the INAV Configurator to the stm32f411ceu6 black battery board. In the Configuration tab, set the I2C Speed to 800 KHZ. Terminate the connection and connect the  of the GY-87 Module wiring to the diagram.
GYRO, ACC, MAG, BARO will work.

Try it and post the result here.

gazza2577, 2021-11-24:
# status
System Uptime: 94 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 8.89V (3S battery - CRITICAL)
CPU Clock=96MHz
STM32 system clocks:
  SYSCLK = 96 MHz
  HCLK   = 96 MHz
  PCLK1  = 48 MHz
  PCLK2  = 96 MHz
Sensor status: GYRO=UNAVAILABLE, ACC=UNAVAILABLE, MAG=UNAVAILABLE, BARO=UNAVAILABLE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE, IMU2=NONE
Stack size: 6144, Stack address: 0x20020000, Heap available: 2048
I2C Errors: 0, config size: 7049, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = none, used = none
   CURRENT : configured = ADC 1, used = none
  AIRSPEED : configured = none, used = none
System load: 13, cycle time: 1006, PID rate: 994, RX rate: 49, System rate: 9
Arming disabled flags: ANGLE CAL HWFAIL RX CLI PWMOUT
VTX: not detected
PWM output init error: Not enough motor outputs/timers

still no joy

gazza2577, 2021-11-24:
# status
System Uptime: 94 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 8.89V (3S battery - CRITICAL)
CPU Clock=96MHz
STM32 system clocks:
SYSCLK = 96 MHz
HCLK = 96 MHz
PCLK1 = 48 MHz
PCLK2 = 96 MHz
Sensor status: GYRO=UNAVAILABLE, ACC=UNAVAILABLE, MAG=UNAVAILABLE,
BARO=UNAVAILABLE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE, IMU2=NONE
Stack size: 6144, Stack address: 0x20020000, Heap available: 2048
I2C Errors: 0, config size: 7049, max available config: 16384
ADC channel usage:
BATTERY : configured = ADC 2, used = ADC 2
RSSI : configured = none, used = none
CURRENT : configured = ADC 1, used = none
AIRSPEED : configured = none, used = none
System load: 13, cycle time: 1006, PID rate: 994, RX rate: 49, System rate:
9
Arming disabled flags: ANGLE CAL HWFAIL RX CLI PWMOUT
VTX: not detected
PWM output init error: Not enough motor outputs/timers

gazza2577, 2021-11-24:
still no joy

rizacelik, 2021-11-24:
Did you follow these steps?

1. First, load the firmware on the STM32F41 board without connecting the modules.
2. Then connect to the board with INAV Configrator.
3. connect the INAV Configurator to the stm32f411ceu6 black battery board.
4. In the Configuration tab, set the I2C Speed to 800 KHZ.
5. Connect the sensor modules to the STM32F41 board.
6. connect the INAV Configurator to the stm32f411ceu6 black battery board.

Have you followed these steps and it didn't work? I can't help you because you haven't given any explanation.

gazza2577, 2021-11-25:
hi done all above dont think the STM32F41 board. i2c is working .
not sure why would changing the i2c pins help like pin 6 and 7

gazza2577, 2021-11-25:
Yes, I did.

rizacelik, 2021-11-25:
With this video, I showed you how to install the sensors. Please watch.

https://youtu.be/HbSUMauSkiw

gazza2577, 2021-11-25:
getting a new board i followed your video to the letter checked sensors are working on arduino
thanks for your help will let you know how i get on

closed 2021-11-26 by gazza2577

gazza2577, 2022-01-31:
hi cant get stm32f411ceu6to work with radio on i bus or s bus
using a Frsky Taranis x9d plus.with fs-ia6b with module plus frsky reciever
using a3 pin rx2 .
do i need to use non-inverted (hacked) sbus
thanks
garry

rizacelik, 2022-01-31:
You can use frsky reciever. The use of SBUS is turned on, you do not need to do anything additional.

gazza2577, 2022-01-31:
got it working on frsky rx6r sbus using a diy sbus inverter .
would not work with sbus from receiver.
thanks for all your help
riza
garry

rizacelik, 2022-01-31:
Great