rizacelik/STM32F411CEU6_INAV_Firmware #3: ExpressLRS or serial base receiver support?

willertan, 2022-06-02:
Thanks for the awesome project! Currently I have expresslrs module on hand and I wish to give it a try. It is serial based TX/RX receiver. may i know where can I wire it?

rizacelik, 2022-06-02:
PA3 PIN -> RX 

You can also use the SOFTSERIAL_1 pins.

SOFTSERIAL_1_TX_PIN     PB6
SOFTSERIAL_1_RX_PIN     PB7

You need to set the receiver protocol for SOFTSERIAL_1 from the control panel.

Agusg666, 2022-06-05:
my elrs on pins A2, A3.
vtx tbs smart audio pin B6 for setting power fvp via oled display

rizacelik, 2022-06-05:
Your explanation is insufficient, I don't understand.

Agusg666, 2022-06-27:
my assembly expresslrs pin rx tx to pA3 and pA2 Full telemetry seen on my
tx fsi6(build erfly6).
i added oled display scl sda to set the power on vtx to my liking 25mw to
500mw to pin PB6(smart audio) and so far no bugs, it's perfect on inav 4.0
. I don't want it on inav 5.0 , thanks to the inav firmware you made.
Greetings from Indonesia to you and your family, may you always be healthy.

closed 2023-10-01 by rizacelik