rizacelik/STM32F411CEU6_INAV_Firmware #4: is it possible to access another UART through B6 and B7

synersignart, 2022-07-05:
since the GPS taken up the TX2RX2  looking that I needed another UART serving as live telemetry

synersignart, 2022-07-19:
also still getting problems of getting the mpu 9250 to work the Gyro, Mag and Acc on occetion the Baro BMP280 goes on and off 
Version 5.0.0

Agusg666, 2022-07-25:
i try set gyro_hardware_lpf = 188HZ, cpu load 17%

synersignart, 2022-07-25:
> i try set gyro_hardware_lpf = 188HZ, cpu load 17%

what would be the CLI command for this?

or any source target code I can look so I  can build the implementation . 

please add
MPU9250
MPU6000
AK8960
MMC5883
thanks

rizacelik, 2022-08-29:
Hi The latest firmware has been recompiled to use the sensors.

closed 2023-10-01 by rizacelik

ahalsaqqa, 2023-10-02:
now the firmware supports 3 UART  ports?

rizacelik, 2023-10-02:
All pins are in use. There are no remaining pins. That's why there is no UART3 support.

ahalsaqqa, 2023-10-02:
https://os.mbed.com/media/uploads/hudakz/blackpill-pinout.png

the UART6 pins not used I think  PA12 and PA11

rizacelik, 2023-10-02:
PA11 and PA12 are used for USB. The system will not work when these two pins are used.