rizacelik/STM32F411CEU6_INAV_Firmware #5: GY91 SPI implementation only GYRO Found

synersignart, 2022-07-31:
I plug the board to SPI1

MPU9250 

Cannot detect in SPI
Mag
Acc
BMP 280

CPU Clock=96MHz, GYRO=MPU6050
STM32 system clocks:
  SYSCLK = 96 MHz
  HCLK   = 96 MHz
  PCLK1  = 48 MHz
  PCLK2  = 96 MHz
Sensor status: GYRO=OK, ACC=UNAVAILABLE, MAG=UNAVAILABLE, BARO=UNAVAILABLE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE, IMU2=NONE

rizacelik, 2022-08-29:
Hi The latest firmware has been recompiled to use the sensors.
I do not have ISP supported sensors, but I will add this support in the next firmware.

synersignart, 2022-09-11:
hmm  can you share the source  as it would be helpful for the INAV community to work bugs out too .  I'll see if i can work this out. also I do need infor on the WS2811 Led strips where did you locate the pins.   also my build now with the help of other community now supports SPI but id good to know that the to have the 6050 available.

Agusg666, 2022-12-06:
Let me share experience using the mpu6050 gyro160 i2c, during the test of more than 100x stable flights never experienced a significant failure while using the inav Rizacelik firmware, one flight failure using inav 5.0 and caused my frame to break, now I'm trying the gy-87 mpu6050 and gy-6500 with cut all sides of the pcb into small pieces and put a soft mount right on the Stm32f411ceu6 development board chips. sorry I use Google translate.

Agusg666, 2022-12-06:
Let me share experience using the mpu6050 gyro160 i2c, during the test of more than 100x flights never experienced a significant failure while using the inav Rizacelik firmware, one flight failure using inav 5.0 and caused my frame to break, now I'm trying the gy-87 mpu6050 and gy-6500 by cutting all sides of the pcb became small and put a soft mount right on the Stm32f411ceu6 development board chips. sorry I use Google translate.

Agusg666, 2022-12-07:
I'm trying to share a channel video with the keyword stm32f411ceu6 INAV gyro pcb cut, sorry I'm not proficient, if you don't like it I'll delete this post

Agusg666, 2022-12-07:
Let me share my experience using the mpu6050 gyro160 i2c, during the test of more than 100x flights I had a significant failure when using the Rizacelik inav firmware, now I tried the gy -87 mpu6050 and gy-6500 by cutting all sides of the PCB into small pieces and installing the soft stand right on the development board chip STM32f411ceu6. sorry I use Google translate.

Agusg666, 2022-12-07:
Let me share my experience using the mpu6050 gyro160 i2c, during the test of more than 100x flights never had a significant failure when using the inav Rizacelik firmware, now I tried the gy -87 mpu6050 and gy-6500 by cutting all sides of the PCB into small pieces and installed the soft stand right on the STM32f411ceu6 chip development board. sorry I use Google translate

Agusg666, 2022-12-07:
Let me share my experience using the mpu6050 gyro160 i2c, during the test of more than 100x flights never had a significant failure when using the inav Rizacelik firmware, now I tried the gy -87 mpu6050 and gy-6500 by cutting all sides of the PCB into small pieces and installed the soft mount right on the STM32f411ceu6 chip development board. sorry I use Google translate

synersignart, 2022-12-19:
ah finally some update but due to chip market inconsistency on the MPU9250 Ive use the BMI160

fabriciosilveira-sjc, 2023-03-20:
Hello @rizacelik. The INAV flash was a success of the STM32F411 (Black Pill).
I'm using the GY-91 module, with MPU 9250 and a BMP280.
In Arduino the GY-91 worked perfectly, but in INAV Accel and Gyro are not recognized, what can I do to solve it.
I tried to use all the fw you made available, but without success.
[image]

rizacelik, 2023-03-23:
Hello;
You have made the cable connections correctly and the BMP280 seems to be working.
In the Configuration tab you should see the MPU-9250 in the sensor list. Select it if it is in the list.

rizacelik, 2023-03-23:
Your link is incorrect. See "Full wire connection" on the main page. Uses SPI Sensors cables B12,B13,B14.
[image]

rizacelik, 2023-03-23:
How is your configuration screen? can I see.

fabriciosilveira-sjc, 2023-03-23:
I'm using version 6.0 of INav.
[image]
[image]
[image]
[image]

rizacelik, 2023-03-23:
[image]
Is the MPU9250 on this list? 
Or what sensors are there?

rizacelik, 2023-03-23:
OK GY-91 itis working I2C. No problem.
INAV SPI connected support.
Please select congigrasyon an test.

[image]

rizacelik, 2023-03-23:
Is your wire connection as follows?

[image]

rizacelik, 2023-03-23:
Looks like the AK8963 magnetometer needs to be bypassed. I will work for it and post the result here.

fabriciosilveira-sjc, 2023-03-23:
Riza, in this GY-91 module, has no magnetometer, only Gyro and Baro.

rizacelik, 2023-03-24:
I did a recompile, can you test it?
[inav_6.0.0_STM32F411CE.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/11060703/inav_6.0.0_STM32F411CE.zip)

fabriciosilveira-sjc, 2023-03-24:
I tried just now and the PC recognized it as DFU, I flashed it through INAV, but the GY-91 still didn't work.

rizacelik, 2023-03-24:
Ok.
MPU9250 is set to I2C1, DFU should work. Please test it.

I2C1

SCL  ->       PB8
SDA   ->      PB9

[inav_6.0.0_STM32F411CE.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/11062886/inav_6.0.0_STM32F411CE.zip)

fabriciosilveira-sjc, 2023-03-24:
I made the connection via I2C.
Barometer okay.
I tested all the Acc options, without success.
[image]

rizacelik, 2023-03-24:
The GY-91 Module is unfortunately seen as MPU9255, not MPU9250. Registers are different.
Necessary warning has been made at this address. https://github.com/ricardozago/GY91-MPU9250-BMP280
I recompiled based on this information, I hope it will be successful this time. Please test and post the result.
[inav_6.0.0_STM32F411CE.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/11064975/inav_6.0.0_STM32F411CE.zip)

fabriciosilveira-sjc, 2023-03-24:
I flashed the new fw.
I tested all MPUs to no avail.
The connection is in I2C.

rizacelik, 2023-03-24:
What's the result? failed?

rizacelik, 2023-03-24:
Can you try for the last time?
[inav_6.0.0_STM32F411CE.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/11065501/inav_6.0.0_STM32F411CE.zip)

fabriciosilveira-sjc, 2023-03-24:
Hello Riza, INAV has the Baro symbol in blue (BMP280).
But the Acc didn't work out.
I've tried every version of INAV you sent me.

rizacelik, 2023-03-24:
I tried a lot but unfortunately it didn't work. It may be chip sourced or the INAV libraries may not be written for it.
I made you test it by changing the registers, but it still didn't work.
Take care of yourselves.

fabriciosilveira-sjc, 2023-03-24:
It worked...!!!!!!!!
https://user-images.githubusercontent.com/73246642/227620307-c4cc3b31-6a8f-41eb-a06a-8b064ad21ff5.mp4

rizacelik, 2023-03-24:
It's great, we did it together. :)
Did you use the latest firmware?
If the latest firmware I sent, I will distribute accordingly.

synersignart, 2023-03-28:
> Riza, in this GY-91 module, has no magnetometer, only Gyro and Baro.
> 
> Sorry so many questions rs

yes oddly the MPU9250 or GY91 has the Gyro and Acc  available in SPI how ever INAV doesnt have the means to get the Mag and Baro in SPI as well in my case I use the seperate Mag and Baro to run i2c

synersignart, 2023-03-28:
> Hello Riza, I can't put Black Pill in DFU on INAV, the computer doesn't recognize it as STM Bootloader. I made the recording of the new fm that you compiled, via FTDI recorder. But it didn't work. I tried to do it in AUTO, MPU6050, MPU6000, MPU6500 and MPU9250, without success. [image] [image]

1st the BackPill  (Weact Studio warns of Clone Black pills that wont flash or operate as intended) pls not of  sources
2nd the STM32F411 has this weird perk that at 25mhz can cause issues getting in DFU mode the work around is to preheat 25c the chip buy plugging it in for a few mins  before plugging it to attempt to DFU mode. (this the reason most FCs that uses this chip runs with a 24mhz crystal) Blackpill is an oddity that runs on 25mhz

synersignart, 2023-03-29:
> Hello sinersignart; There are three different MPU9250 modules on the market. WHOMI records are 0x71 , 0x73, and 0x75. INAV only uses the real WHOMI record with 0x71. Other modules are column products. Let's test with you as we did with mr fabriciosilveira-sjc. I recompiled according to the FIRMWARE 0x71 , 0x73, and 0x75 values I will post below. We can get around this problem if you test it and report the result. [inav_6.0.0_STM32F411CE.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/11086694/inav_6.0.0_STM32F411CE.zip)

I see , which one was adjusted in the the drivers accgyro_mpu9250.c

synersignart, 2023-03-29:
where does the file IMU registries are stored in INAV
?

synersignart, 2023-03-30:
> It's in the gyro_mpu.h file. Did the Firmware I sent sinersignart work?

ah the reason id ask is that  im using our board which has very different Pin assignments hence I required to rebuild the INAV 

so I just change  `#define MPU9250_WHO_AM_I_CONST              (0x71) `

the `#define MPU9255_WHO_AM_I_CONST              (0x73)` might conflict from INAV default?

rizacelik, 2023-03-30:
Your English is not fully understandable. No, conflict from INAV default.

fabriciosilveira-sjc, 2023-06-20:
Hello Riza... once again many thanks for your help with connection and firmware for the GY-91.
Another question, does the HC-SR04 signal work with the STM32F411 (Black pill) running INAV 6?

fabriciosilveira-sjc, 2023-06-20:
> Olá. Sory, preterido, não funcionará com INAV 2.0 e posterior.

OK.
I have two questions:

1. Does STM32F103 (Blue Pill) run INAV 1.7.3? I couldn't make it work. That's for me to try using sonar.

2. Is there a compatible sonar for A STM32F411 (Black Pill)?

Lalbabu2001, 2023-09-01:
> I tried just now and the PC recognized it as DFU, I flashed it through INAV, but the GY-91 still didn't work.

How did you manage to get back dfu mode mine has same problem after flashing this crappy firmware.

closed 2023-10-01 by rizacelik