rizacelik/STM32F411CEU6_INAV_Firmware #6: WeAct BlackPill STM32F411CEU6 fake blackpill?

willertan, 2022-10-04:
I am not sure is this firmware problem or hardware problem. My first build using this STM32F411CEU6 Development Board ( https://www.aliexpress.com/i/4001251992131.html ) was success. and my 2nd build with WeAct STM32F411CEU6 ( https://www.dfrobot.com/product-2338.html ) was fail in both iNAV and cube programmer.
Can anyone confirm this is hardware problem before I talk to seller?
Thanks

-----------------------CUBE PROGRAMMER-------------------------
10:48:45 : USB speed   : Full Speed (12MBit/s)
  10:48:45 : Manuf. ID   : STMicroelectronics
  10:48:45 : Product ID  : STM32  BOOTLOADER
  10:48:45 : SN          : 368D345C3535
  10:48:45 : DFU protocol: 1.1
  10:48:45 : Board       : --
  10:48:45 : Device ID   : 0x0431
  10:48:45 : UPLOADING OPTION BYTES DATA ...
  10:48:45 :   Bank          : 0x00
  10:48:45 :   Address       : 0x1fffc000
  10:48:45 :   Size          : 16 Bytes
  10:48:45 : UPLOADING ...
  10:48:45 :   Size          : 1024 Bytes
  10:48:45 :   Address       : 0x8000000
  10:48:45 : Read progress:
  10:48:45 : Data read successfully
  10:48:45 : Time elapsed during the read operation is: 00:00:00.042
  10:50:28 : Read File: C:\Users\uuu\Downloads\inav_5.1.0_STM32F411CE_Upgrade.hex
  10:50:28 : Number of segments: 2
  10:50:28 : segment[0]: address= 0x8000000, size= 0x400
  10:50:28 : segment[1]: address= 0x8008000, size= 0x6DDD4
  10:50:54 : Memory Programming ...
  10:50:54 : Opening and parsing file: inav_5.1.0_STM32F411CE_Upgrade.hex
  10:50:54 :   File          : inav_5.1.0_STM32F411CE_Upgrade.hex
  10:50:54 :   Size          : 440.46 KB 
  10:50:54 :   Address       : 0x08000000 
  10:50:54 : Erasing memory corresponding to segment 0:
  10:50:54 : Erasing internal memory sector 0
  10:50:54 : erasing sector 0000 @: 0x08000000 done
  10:50:54 : Erasing memory corresponding to segment 1:
  10:50:54 : Erasing internal memory sectors [2 7]
  10:50:55 : erasing sector 0002 @: 0x08008000 done
  10:50:55 : erasing sector 0003 @: 0x0800c000 done
  10:50:56 : erasing sector 0004 @: 0x08010000 done
  10:50:58 : erasing sector 0005 @: 0x08020000 done
  10:51:00 : erasing sector 0006 @: 0x08040000 done
  10:51:02 : erasing sector 0007 @: 0x08060000 done
  10:51:02 : Download in Progress:
  10:51:02 : Error: Write failed at address: 0x8008800 
  10:51:02 : Error: failed to download Segment[1]
  10:51:02 : Error: failed to download the File

[image]
[image]
[image]
[image]

rizacelik, 2022-10-04:
Did you use INAV Configurator 5.1? It looks like there is a chrome extension in the images.
Try it by downloading INAV Configurator 5.1 from https://github.com/iNavFlight/inav-configurator/releases.
Note: Try the version with inav_5.1.0_STM32F411CE.hex

willertan, 2022-10-10:
> Did you use INAV Configurator 5.1? It looks like there is a chrome extension in the images. Try it by downloading INAV Configurator 5.1 from https://github.com/iNavFlight/inav-configurator/releases. Note: Try the version with inav_5.1.0_STM32F411CE.hex

I had try every combination of INAV Configurator (program), STM cube programmer, Arduino STM32 sdk, with all iNav version. erasing is ok but fail at writing in firmware. I believe my black pill board is fake or rebranded STM chip. I will do another test using usb st-link dongle to swsdk/swdio pin and update u guys here. thanks

synersignart, 2022-12-19:
I have to say there are alot of Fack Blackpill on the market that wont work reliably  which several of them wont even flash the fimware correctly.  the Question is if you got it from the Official site?

Vinnikau, 2023-01-20:
Hi! I catch this problem too. I bought blackPill not in WeAct store. This board succcess erased, but when inav configurator try to program it stopped with message "Programming: FAILED".

Maybe add this problem in description in read.me?

Vinnikau, 2023-01-20:
Also i fixed this problem. I erase fake board by st-link. I use [inav_6.0.0_STM32F411CE.hex](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/blob/main/inav_6.0.0_STM32F411CE.hex) and SMT32 St-link utility.

But I think not all fake boards can be erase.

rizacelik, 2023-01-20:
Did you use INAV Configurator 6? https://github.com/iNavFlight/inav-configurator/releases

Vinnikau, 2023-01-20:
> Did you use INAV Configurator 6? https://github.com/iNavFlight/inav-configurator/releases

First i used INAV Configurator 6 and [inav_6.0.0_STM32F411CE.hex](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/blob/main/inav_6.0.0_STM32F411CE.hex)
Next i used INAV Configurator 5.1 and [inav_5.1.0_STM32F411CE.hex](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/blob/main/inav_5.1.0_STM32F411CE.hex)

And Last I programmed my board in SMT32 St-link utility and [inav_6.0.0_STM32F411CE.hex](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/blob/main/inav_6.0.0_STM32F411CE.hex). 

Now for configuration I use INAV Configurator 6 (HW-290 sensors board) it works perfect.

rizacelik, 2023-01-20:
Wonderful

synersignart, 2023-01-21:
I manage to have one fake one working but the issue is it can be inconsistent but guess I just got lucky.  other users may run into  difficulties

closed 2023-10-01 by rizacelik