rizacelik/STM32F411CEU6_INAV_Firmware #8: it is possible to remap motor and servo outputs?

ASADAIDAR, 2022-10-18:
I assembled a budget flight controller for a flying wing according to the instructions (3 outputs: 2 servos and 1 motor), but for convenience I need to remap the outputs S1 to pin B3; S2 to pin A8; S3 to pin B4;S6 to pin B10. Could you compile a new firmware with these outputs?I made handy pins (pictured) to connect the servos and motors, so I need to remap the outputs.
[image]
[image]

rizacelik, 2022-10-20:
Unfortunately I don't have time for this. 
Sory.

rizacelik, 2022-10-24:
Everything you will need has been made, why not consider doing it with a few cables. The easiest way is to combine a PCB with a board.

closed 2023-10-01 by rizacelik