rizacelik/STM32F411CEU6_INAV_Firmware #9: Source location?

sensei-hacker, 2022-11-04:
Where is the source location, please?
You're actually violating the copyright by not having that information readily visible, such is in the README.

rizacelik, 2022-11-05:
There is nothing hidden. How to do it is described in the documentation here https://github.com/iNavFlight/inav/tree/master/docs/development. I am doing it according to the instructions here.
In addition, it is not a commercial transaction, anyone can download and use it freely. Why did this bother you so much?
I used https://github.com/iNavFlight/inav/tree/master/src/main/target/MATEKF411TE as source board.
I hope your inconvenience has been resolved.

sensei-hacker, 2022-11-05:
> How to do it is described in the documentation here

I'm not asking you how to write an inav target. I know quite well how to write a target. I'm asking you to stop stealing our work - to follow the license by publishing the changes you made to our target.

Yes, I can see you copied the Matek F11TE target. Which is the same as if you made copies of someone's song or book - you can do so only as allowed by their license.

rizacelik, 2022-11-05:
INAV
GNU General Public License v3.0

Commercial use
 Modification
 Distribution
 Patent use
 Private use

The INAV source is as follows.
Multiwii -> CleanFlight -> Betaflight -> INAV

My work is not source code distribution. My work is to produce firmware for a board using INAV source codes. Who gives you the right to say "an insolent word to steal your work"?

sensei-hacker, 2022-11-05:
I will try to explain this to you briefly. We release our inav work under GPL so that you, Michael, and anyone else can use and modify it.

You've taken that and tried to away Michael's right to modify it further.

What you've been doing is akin to walking into a public park, on land donated for a public park, and proclaiming "this is now my private property". That's not okay. It's not your property! It's not up to you to make the rules.

The rules are, share and share alike. If you refuse to respect Michael's right to further modify the code, you are not within any license and you have no right to use or distribute the software at all.

You can EITHER:

1) Ise and distribute the software in compliance with the license
2) Not use and distribute it at all

Inav is NOT your private property. You don't get to copy out work and declare your keeping it to yourself.

rizacelik, 2022-11-05:
Why don't you want to understand?
1. My work is firmware like commercial cards in target in INAV.
2. The license is already licensed under INAV and compiled under this license.
3. I don't have any explanation for "I have this". Anyone can take it and use it. Just like any other firmware.
4. Who are you and what is your authority? Let the person in charge contact me.

sensei-hacker, 2022-11-05:
Read the license for inav.

The copies you are distributing are NOT licensed, because you are refusing to accept the only available license.

Inav is only available under the GPL license, which requires that you share your modified source. If you don't do that, you aren't following the GPL license and you have no license at all.

> Who are you and what is your authority? Let the person in charge contact me.

I'm one of the copyright holders - you're stealing my work.
I'm currently filling out a DMCA notice to have your repo removed unless and until you comply with the license.

rizacelik, 2022-11-05:
GPL License, in short.

Unlimited freedom to use the program.
Freedom to understand how the program works and that it can be modified for specific purposes.
Freedom of unrestricted distribution of copies of the program.
Freedom to distribute modified version of the program.

The scope of the license is as above.
You can apply wherever you want. You are free in this.

sensei-hacker, 2022-11-05:
See section 6 of the license:

--
You may convey a covered work in object code form under the terms
of sections 4 and 5, provided that you also convey the
machine-readable Corresponding Source under the terms of this License
--

Why?  Because as the license states:
--
the GNU General Public License is intended to guarantee your freedom to
share and change all versions of a program--to make sure it remains free
software for all its users.
--

You are trying to take away that freedom from others. You aren't allowed to do that, while still taking our work. You can EITHER get inav, OR make your proprietary software. You can't make our work proprietary.

rizacelik, 2022-11-05:
`"Hex dosyalarını YALNIZCA ilgili kaynağı da dağıtırsanız dağıtabilirsiniz."`

the source of interest is already in the INAV main repository, isn't it silly to move them all here?

I will put the link of the INAV main repository in the README file. I will put the developer link as well. All the details on how to do it are already there. I just prepared the target.h and target.c files, the rest is the INAV source.

sensei-hacker, 2022-11-05:
> isn't it silly to move them all here?

I would tend to agree. Linking to the main inav repo *is* providing *most* of the files.

I don't think anyone would complain if you post only the files that aren't in the main repo - the files you changed.

Thanks.

rizacelik, 2022-11-05:
Anyone can complain wherever they want, no problem for me. I'm not making any money off of this. I have no intention of becoming famous by doing this. I did it to do a different study, that's all.

closed 2022-11-05 by rizacelik