rizacelik/STM32F411CEU6_INAV_Firmware #12: FrSky receiver

fabriciosilveira-sjc, 2023-06-20:
Hello Riza...

I managed to make the STM32F411 work with Gyro, Acc, Baro and GPS, wow... thanks for the help.

Now I'm installing a FrSky receiver, model XSR.
I installed it as S.BUS on pin A3.
In INAV, in "Ports" I enabled Serial RX on UART 2, in "Receiver", I selected Serial > Sbus. Then I save and restart, and the receiver is not recognized.
I have tested the receiver on the ToolkitRC charger and it works perfectly.

[image]
[image]

rizacelik, 2023-06-24:
Check the wire connection. It should work.

Lalbabu2001, 2023-09-01:
> Hello Riza...
>
> I managed to make the STM32F411 work with Gyro, Acc, Baro and GPS, wow... thanks for the help.
>
> Now I'm installing a FrSky receiver, model XSR. I installed it as S.BUS on pin A3. In [...]

Hey did you manages to subs work?

closed 2023-10-01 by rizacelik