rizacelik/STM32F411CEU6_INAV_Firmware #13: i installed 6.1.1 but i can't connect, it says MSP connection successfully opened with ID: 1 No configuration received within 10 seconds, communication failed MSP connection successfully closed

DONGHUUTHANG, 2023-06-22:


longlong7211, 2023-07-07:
I disconnected all but it did not work, the same issue with this topic. I try flash 6.1.0 and 6.1.1 it does not work but 6.0.0 it work

longlong7211, 2023-07-09:
it is not stable for me, sometimes I can connect but setup and reconnect it does not work, and it same issue for me

rizacelik, 2023-07-09:
Your USB TYPE-C Cable may not be suitable. You must purchase one TYPE-C data cable. Charging cables are useless.
Tested with multiple STM32F411CEU6 boards, working.

longlong7211, 2023-07-24:
no, led 13 doesn't blink when connecting power. I don't connect mpu, cpu load 20% but connect mpu it overload, it always over 100%

kofbox, 2023-07-25:
Hello!

I did some research on this issue, as this firmware is also very interesting.

So, I tried:
- three firmware versions: inav_6.0.0_STM32F411CE.hex, inav_6.1.0_STM32F411CE.hex, inav_6.1.1_STM32F411CE.hex
- two different laptops, and one stationary computer, in order to avoid a lack of voltage on the USB port
- 4 different USB-A to USB-C cables, 1 USB-C to USB-C cable
- OS Windows 11, Linux Ubuntu 22.04 (with version INAV Configurator 6.1)
- two different BlackPill options - with 8MFlash and without

In all the above options, this firmware does not work normally.

It should be said that BlackPill was purchased from the manufacturer's official store on Aliexpress. Version - 3.1

As for the USB cable: I originally used two inexpensive USB-A to USB-C cables and one USB-C to USB-C. After reading this message thread, I followed the advice of @rizacelik and bought two fairly branded and expensive USB-A to USB-C cables (also quite thick as it turned out) - HP USB3.1 (model DHC-TC101-1M) and Duracell USB3.2Gen1 (this one is the most expensive ~18 euros). With these cables, the firmware and the board behaved in the same way as with other cheaper ones.

Additional research:

Every time you connect a BlackPill with this firmware, the blue led is constantly on without blinking. However, once (accidentally) I touched the BlackPill with my finger to pins B13 and B14 for 1 second and the blue led went out. I intentionally shorted these two pins for another second and the blue led blinked - an indication of normal loading and regular operation of this firmware (judging by the @rizacelik video on YouTube).

Next, I took a jumper and deliberately connected pins B13 and B14 and turned on the BlackPill to the USB cable. And a miracle happened, the firmware came to life, INAV successfully identified it and allowed it to enter and configure.

With this jumper, I successfully connected the HMC5883, BMP280, MPU6050 sensors. The configurator normally displayed the data received from the sensors and communicated with them. However, when it came to connecting the ESC and the motor (with and without this jumper, but shorting pins B13 and B14 twice), the ESC did not receive a signal to start the motor. At the same time, a sound was heard from the motor indicating that the ESC was connected normally and ready for control (in the future it turned out that this indication was much quieter than when BlackPill was connected with firmware from other developers).

When flashing BlackPill with firmware from other developers - there are no such problems - all five USB cables work, the INAV configurator works with any OS, both versions of BlackPill (with 8MFlash and without) are flashed, and it also works from all tested computers / laptops. However, these firmware are either outdated and not supported by developers, or do not have such a set of connected sensors, which is why I spent so much time / resources / money in order to try to understand this problem and state it all here, so that, dear @rizacelik, for his part, could fix this problem.

I will be very grateful for this fix, and for my part I am ready to provide any additional information, as well as remote access to any configuration of the current equipment and available BlackPill / USB cable / OS, and I am also ready to contact in the messenger and be "hands here" if this method is needed.

I am also attaching a photo of the jumper on the BlackPill. Let me know, if necessary - I will record a video of these manipulations - for a good example.

I'm also sure that @longlong7211 has the same problem, and if you try to put a jumper, you will be able to enter the configurator and partially work the firmware, as in my example.

[https://pasteboard.co/8LHAKuuaKBvr.jpg](https://pasteboard.co/8LHAKuuaKBvr.jpg)
[https://pasteboard.co/YtICkldxpBq6.jpg](https://pasteboard.co/YtICkldxpBq6.jpg)

rizacelik, 2023-07-25:
Hi @kofbox
First of all, thank you for the feedback.

PB13 and PB14 pins belong to the SPI interface and these pins can expect signals from OSD, MPU6500 or MPU9250 sensors. It may be causing problems especially since there is no built-in OSD.
It doesn't work on my microcontroller. But you and a few others have reported problems. Your shorting of the PB13 and PB14 pins indicates that the problem is in the SPI interface.
I recompiled the firmware by disabling the OSD in the SPI interface. I would appreciate if you test it and let me know the result.
[inav_6.1.1_STM32F411CE.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/12165498/inav_6.1.1_STM32F411CE.zip)

kofbox, 2023-07-25:
Hello, and thanks for responding! I'll be happy to run tests.

The board came to life without a jumper with this fix!

The servo started working normally (before the fix it didn't work).

The motor and ESC make a sound of readiness / connection to the controller, however, when adding revolutions in the INAV configurator, nothing happens, the motor does not rotate (the same situation as before the fix). Motor connection to pins - A8 and Ground. The same situation when connected to A9.

At the same time, the same motor and ESC on firmware from another developer behaves normally.

Additionally, I can say that the "ESC readiness sound" on all @rizacelik firmwares is much quieter than on firmwares from other developers (the same situation as before the fix). Pretty weird...

Next are the cardinal differences from the previous firmware version inav_6.1.1_STM32F411CE.hex:

HMC5883, BMP280, MPU6050 sensors are connected and detected normally. At the same time, the processor is loaded by ~ 60%. On the previous firmware, the load on the processor with the same sensors (and the jumper inserted) is ~ 6%.

If you need any additional tests - let me know, I'll be happy to perform them.

P.S.: An additional question - will only the OSD be unavailable on the SPI interface, or is the entire SPI interface now disabled?

rizacelik, 2023-07-25:
Thank you for the test. We will work together to solve the problems.

It's nice that it works fine after this minor fix.
CPU load should normally be around 10%, but because the MPU6050 sensor works very slowly on I2C, INAV has removed this support, but I added it again.
If you want a stable flight, definitely connect it to the MPU6500 SPI interface. CPU load will be much lower.

The SPI interface is working, not disabled, just the OSD has been removed from the default.

ESC configuration must be done for ESC and motors to work. What is your ESC brand?

I will be glad if you answer.

rizacelik, 2023-07-25:
If you do not have the MPU6500, but you want to work with the MPU6050, you can reduce the CPU load by typing the following commands on the command line in the CLI interface to reduce the CPU load.

set looptime = 1000
set gyro_hardware_lpf = 188HZ

These two values will be useful to you.

kofbox, 2023-07-25:
I am using a non-name ESC and a motor. But they also work fine on the same BlackPill but with firmware from another developer. I am attaching links to the pages where I bought them.

Also about the load on the processor - how to explain that on the firmware before the fix, the load on the processor is only ~ 6%, and with the fix ~ 60%? While the firmware settings are the same.

[ESC](https://www.aliexpress.com/item/1005005032261137.html)
[Motor](https://www.aliexpress.com/item/4001177046489.html)
[Photo](https://pasteboard.co/CbAFPSWzw1NY.jpg)

rizacelik, 2023-07-25:
[image]
ESC you should use a normal ESC and STANDARD settings.
I also disabled the settings related to CPU load, so you got 60%.
I wrote the commands I disabled in my previous reply. Use them CPU load will drop.

kofbox, 2023-07-25:
Ok, I can try them tommorow and whrite to you. Thanks!

kofbox, 2023-07-26:
With your commands - the load on the processor has dropped to ~ 7%, thanks, I will use them.

Regarding the ESC settings in more detail (the same situation on the firmware before the fix and with fix):

STANDARD - no ESC ready sound, motor does not rotate
ONESHOT125 - ESC ready sound, motor does not rotate
MULTISHOT - there is an ESC readiness sound, the motor rotates, but not normally, starts at 28%, stops rotating at maximum speed at 89%
BRUSHED - initially there is no ESC ready sound, but when you try to increase the speed from 0% to 50%, the ESC ready sound appears, but the motor does not rotate
DSHOT150 - ESC ready sound, motor does not rotate
DSHOT300 - ESC ready sound, motor does not rotate
DSHOT600 - ESC ready sound, motor does not rotate

For comparison - on the firmware from another developer - STANDARD - ESC ready sound, the motor rotates, starts at 1%, rotates at maximum speed by 100%.

I can give you remote access to a laptop with a BlackPill connected for a more detailed review if necessary.

rizacelik, 2023-07-26:
Hi @kofbox.

MULTISHOT seems suitable for ESC. Calibrating your ESC in iNav. Watch this video https://www.youtube.com/watch?v=1IrgbY0YhqM and if the problem still persists I will recompile version 6.1.1 according to the settings in INAV version 5.0.
It looks like it's probably a Calibration issue.

rizacelik, 2023-07-26:
If the problem persists, if possible, you can take a video of the ESC calibration and upload it here. It will be more understandable.

kofbox, 2023-07-26:
At the moment, on the firmware with the fix, I caught an incomprehensible bug that was previously there, but not so constantly (apparently due to the fact that before doing ESC, I configured other parameters in the firmware). At the moment - when flashing and setting up ESC - BlackPill is constantly disconnected from the configurator. And I don’t understand - what was my previous sequence of actions, that I avoided such a bug ...

On the firmware before the fix, as well as on the firmware of another developer, there is no such problem, which indicates that this is not a problem of the hardware level of the board.

I am attaching a video that shows step by step what actions lead to the problem. In the video, I flash with a firmware with a fix and observe this bug, then I flash the firmware of another developer and this bug is not there, and then I flash it again with a firmware with a fix and this bug is present.

[https://youtu.be/Cn2Mpm998TA](https://youtu.be/Cn2Mpm998TA)

kofbox, 2023-07-26:
Ok, while the symptom progresses on firmware without a fix. Now I think it might be some kind of hardware problem in the board...

rizacelik, 2023-07-26:
[image]

If I'm not mistaken in the video, you choose "keep current setting". This is not recommended and will throw an error.

Do not select this option. Choose one of the other two options for the plane that suits you.

Can you test it one more time?

kofbox, 2023-07-27:
I confirm that the problem has worsened, now with the same actions the board is turned off and it is no longer possible to connect it until a complete flashing

I also checked the option without using the "Keep current settings" in menu, the problem is the same.

 It looks like the board itself is dying, this is the second one. The first one died when the (apparently) MPU6500 was connected incorrectly. At the moment, I have the last (third) board with 8Mflash left. I will continue testing on it while I order a few more from Aliexpress and expect them in a month.

@rizacelik  I would like to ask, since I can order a variant with or without 8Mflash, is there any benefit in having this memory? Can I somehow additionally use it with this firmware?

rizacelik, 2023-07-28:
STM32F411CEU6 100Mhz, 128Kb RAM, 512Kb ROM; 25Mhz high speed crystal oscillator.

8Mflash ?   I couldn't understand 8Mflash. Do you mean 8 Mhz crystal oscillator?

The firmware runs only on the board with a 25 Mhz crystal oscillator.

kofbox, 2023-07-28:
I buy the original from the manufacturer (I think so) and the product page has an option with and without 8Mflash.

https://www.aliexpress.com/item/1005001456186625.html

rizacelik, 2023-07-28:
I looked into the details but I couldn't see what 8Mflash does. The price is higher than the other.
If it's not a problem for you, get 1 regular one and 1 8Mflash one.
Since the product detail is very insufficient, I cannot say anything about it, but I would order the normal one.
If you can ask the manufacturer a question, ask what 8Mflash means.

rizacelik, 2023-07-28:
8Mflash meaning. (8 Mb SPI Flash)

SPI (Serial Peripheral Interface) NAND Flash provides an ultra cost-effective while high density non-volatile memory storage solution for embedded systems, based on an industry-standard NAND Flash memory core.

kofbox, 2023-07-28:
Based on this description - maybe this version is useful for this firmware?

I can also say that this is the only working version I have now. Therefore, I can give you the opportunity now to consider it in more detail, if necessary.

Based on the fact that two pieces are already damaged, while the second one stopped working normally not from an incorrect connection, but by itself - I see a trend that I need to buy more than two pieces, since it is not clear how many more can go bad before all the firmware adjustments are completed.

rizacelik, 2023-07-28:
8 Mb SPI Flash not required for INAV. It may be necessary for data logging for different applications. So you can get the standard one.
Use what you have carefully, avoiding short-circuit and incorrect +, - power use, no need to spend too much money.

kofbox, 2023-07-28:
Good day to you.

So:
- I soldered the pins on the third board
- flashed it with firmware with a fix
- when connecting to the configurator for the first time, I chose the option "Airplane with a Tail"
- applied your commands in the CLI to reduce the load on the processor, saved the parameters and rebooted
- in the Outputs menu, enabled the "Enable motor and servo output" parameter, saved the settings and rebooted the board

Further, according to the instructions from the video, before applying power to the ESC, in the configurator, I raised the engine speed to the maximum, turned on the ESC with the motor, the motor squeaked a greeting, but there was no sound that the ESC started calibration, I waited a couple of seconds, lowered the speed to 0%. After that I tried to increase the speed, but nothing happened.

I tried the calibration procedure on another firmware, ESC gave out additional sounds there, but there are too many of them, whole melodies. And the calibration seemed to be non-standard - 100% -0% -100% -0%. But I'm not sure if this ESC supports calibration at all. Since it is very cheap and it does not even have the ability to adjust the parameters, they are preset by the manufacturer and are not saved when changed.

If you advise any proven 10-20A ESC, then I will order it and test it with it.

However, at least on another firmware, the current ESC behaves differently than on the current one and after the procedures I described - it starts the motor (as well as without calibration). The motor also starts on the current firmware, without calibration in MULTISHOT mode, but not from 0% to 100%, but from 28% to 89% (as I wrote in the previous comment). I tried to calibrate in this mode, but it also failed, ESC continued to work within 28% - 89% in this mode.

rizacelik, 2023-07-29:
You have Simonk brand ESC. For calibration, first watch https://www.youtube.com/watch?v=yaAf4eofppc video, then https://www.youtube.com/watch?v=c55DrR-vngE this video. If it doesn't help then I will advise you to buy a new one.

You can also re-flash your ESC (simonk esc firmware update) if you want. For this, you can search from sources such as google or youtube.

kofbox, 2023-07-29:
I suspect that this is not branded, but a replica, and without the ability to change parameters and calibrate.

With your firmware in STANDARD ESC mode, it only produces a welcome sound. There are no additional sounds.

With another firmware - there are too many sounds, I am attaching an example [video](https://youtu.be/VdSX-vEtwOA) (sorry that the ESC sound is quiet - there is no other, larger motor at hand). At the same time, no tuning examples on the Internet match the current sounds. I've watched the videos you suggested before and can't get a similar result.

Regarding flashing the ESC - I studied this issue and I'm afraid that my knowledge is not enough to properly flash the ESC. For example, based on this [video](https://www.youtube.com/watch?v=MQQX7IIdAtI).

In addition, I asked the seller a question about whether this ESC can be configured and whether there is any kind of manual - the answer is no.

rizacelik, 2023-07-30:
I understand.
With another firmware you can use - your ESC is working fine.
Why don't you use another firmware?

I have used this ESC a lot and had no problems.
https://www.aliexpress.us/item/2251832654074337.html?gatewayAdapt=4itemAdapt

kofbox, 2023-07-30:
> With another firmware you can use - your ESC is working fine.

I wouldn't say so. It seems that with a different firmware - this ESC just works, not good, but just works. I still can't calibrate it, change the parameters in it, and I'm going to use folding screws, and for this it is extremely necessary to be able to adjust the motor braking. I don't know - I bought this ESC, because I thought that they were all configurable. And the seller in the description did not indicate the fact that these ESCs are locked for configuration and calibration.

> Why don't you use another firmware?

I have several reasons:

- I wanted to use the GY-87 board, since it combines all the sensors, which means it saves weight and size, but here the MPU-6050 accelerometer sensor that you returned to the firmware, unlike other developers.

- Your firmware is well documented, which clearly affects the process, since I'm new to this, but I want to figure out exactly what works and how, and not look for easy ways. I want to understand why something works or does not work.

- Firmwares from other developers become obsolete and are not kept up to date, in addition, they contain a much smaller number of sensors, since the developers made these firmware "for themselves", and therefore for their set of sensors.

> I have used this ESC a lot and had no problems.

Delivery of this product to my country costs the same (for some reason) as the ESC itself. I found a [similar product](https://www.aliexpress.com/item/1005001621838450.html), with the same description, which says that it can be adjusted and calibrated, but the shipping is much cheaper. I will order it and (in about a month) continue with it. So far I only have the current ESC.

Thank you for your participation in resolving issues. If I can do something else at the moment to improve the firmware, I will be happy to help.

Other than that, I'm looking for a way to use an FPV camera with OSD telemetry, but I can't figure out how. You originally wrote that you disabled the OSD in this firmware. Perhaps you can suggest a workaround, or can I help test the OSD to get it working?

Lalbabu2001, 2023-09-01:
> I wouldn't say so. It seems that with a different firmware - this ESC just works, not good, but just works. I still can't calibrate it, change the parameters in it, and I'm going to use folding [...]

Hey buddy do you know about other developer firmware may you provide me links of another firmware links?

closed 2023-10-01 by rizacelik