rizacelik/STM32F411CEU6_INAV_Firmware #14: I installed 6.1 firmware, having trouble in controlling motors with my rf controller

paky12, 2023-06-23:
Hello sir, im having, troubles with rf communication. In inav everything works fine, joystick controls are visible (using Ibus), motors can be controlled from inav and gyro is calibrated fine. Problem is when i try to control the drone using my rf controller. When i start giving throttle at certain point it goes to maximum and wont turn off, also the other joystick wont work. Wondering if you could provide me with source code of firmware perhaps on my email : patrikdrazic@gmail.com

rizacelik, 2023-06-23:
Hello Patrick.
Did you set the ESC protocol correctly? The ESC protocol should be MULTISHOT.
Source codes are target.h and target.c files. You can review. You can compile it yourself if you want. See description section.

rizacelik, 2023-06-23:
Did you enable ARM from the Modes section in the configurator?

paky12, 2023-06-23:
So i have made all the steps like you, what happens is i ARM the drone but when i start giving throttle it jumps to maximum speed and i lose control. I think it looses connection with my RF transmitter and wont turn off, i have to plug the batteries out to stop the motors. Here is the short video demonstrating how INAV sees everything https://streamable.com/34hq1f

rizacelik, 2023-06-23:
Is there a brand of ESC?
How are your mixer settings?
Also try different ESC protocols.

rizacelik, 2023-06-23:
Whichever channel you have completed for ARM on the transmitter, you need to activate it on that channel. It seems that ARM is not activated from the transmitter.

paky12, 2023-06-23:
BLHeli-s 2-4s it supports multiple protocols, always the same problem.

rizacelik, 2023-06-23:
Your BLHeli-s ESC settings may not be exact. Someone else asked on youtube and I recommended the following videos.
You can also try other protocols. These videos can help you.
https://www.youtube.com/watch?v=xf9Jhoooxec
https://www.youtube.com/watch?v=8YY9BTsa6MQ

When you find a solution, please reply here again about the solution.

rizacelik, 2023-06-23:
Try inav_6.0.0_STM32F411CE.hex Firmware. If it works on that then we can make changes for 6.1.

INAV flight software is very complex and requires a lot of programming knowledge. Making changes to the software will take a good programmer months. So it is very difficult for me to write a block diagram of program flow with logical sequences for drone software. There are so many details.

paky12, 2023-06-24:
Understandable, thank you for fast response

rizacelik, 2023-06-24:
You wrote that you didn't stop your engines.
I tested it today. It worked flawlessly. When you activate the ARM channel, the engines will run. Engines will stop when you disable the ARM channel. After ARM you are ready to fly. This is a security measure.

paky12, 2023-06-25:
Yes the ARM channel activates the engines but when i give throttle it goes to maximum value by itself and i cannot control it (i had to make also a KILLSWITCH channel just so i can shut it off in time before making any more damage), might be that I have to do some PID tuning

rizacelik, 2023-06-26:
If you are using mpu9250 it works unstable. There can be high ups and downs. Use mpu6500 instead.

paky12, 2023-06-26:
Ahh yess im using mpu 6050 with I2c, tho with some PID tuning i managed to get better stability but still... wont be able to fly

rizacelik, 2023-06-26:
Why doesn't it fly? what exactly is the problem? I still do not understand.

paky12, 2023-06-26:
Cpu load is okay also under 80%, professor told me today that if i understood correctly something about PID is incorrect, should  i use default settings? PRoblem is for exxample in Angle mode it pushes motors in wrong direction, it should try to balance out the drone so it stays flat but it goes in different direction. It tries to unbalance the drone so it can't even lift off, it dosent give power to other side of motors to provide balance. I dont know how to explain im sorry

paky12, 2023-06-26:
Here is a short video of how it looks, the channel i was switching on and off while mottors where spinning is ANGLE mode.
https://streamable.com/tzxfcu

rizacelik, 2023-06-26:
You can be flown, the PID controller will stabilize your quadcopter. I recommend you to use it with MPU6500 SPI.
Note: The MPU6050 I2C connection is overworking the CPU. So the best is the MPU6500 module. I'm using this too.

rizacelik, 2023-06-26:
Since Gayro&Acell modules increase CPU load too much in I2C, INAV is no longer supported. The MPU6500 module is very cheap and you can find it easily.

Lalbabu2001, 2023-06-26:
Hey, let me join this discussion, i too have same problem, the problem is neither motors not esc, its the firmware who is brutal,  when you look at the output tab you will realise, even i tried to manage lift off but the motors will never be behave like it should, when throttle down one of the motors still gets 70% signal, the quad is levelled and the pitch roll and yaw are on 1500 the what is the cause??

Lalbabu2001, 2023-06-26:
> Hello Patrick. Did you set the ESC protocol correctly? The ESC protocol should be MULTISHOT. Source codes are target.h and target.c files. You can review. You can compile it yourself if you want. See description section.

Please help

rizacelik, 2023-06-26:
SOLUTION:
 I recommend you to use it with MPU6500 SPI.
Note: The MPU6050 I2C connection is overworking the CPU. So the best is the MPU6500 module. I'm using this too.
OR
Test firmware 6.0, do you have the same issue in that version?

rizacelik, 2023-06-26:
If the ESC calibration is not done properly, the motors do not rotate at the same speeds simultaneously. For this, you need to do the ESC calibration first. You can find lots of videos on Youtube.

paky12, 2023-06-26:
I think i had same problems with 6.0 ill try again

rizacelik, 2023-06-26:
Although all the installation steps are done properly, if there are problems, we can fix it together.
  I'm just going to do a basic build for the quadcopter. Can you test this? (Can all of the friends participating in the discussion do this?).
Feedback is important for the solution.

Lalbabu2001, 2023-06-26:
> SOLUTION: I recommend you to use it with MPU6500 SPI. Note: The MPU6050 I2C connection is overworking the CPU. So the best is the MPU6500 module. I'm using this too. OR Test firmware 6.0, do you have the same issue in that version?

Yes, both the versions have same problem i tested both inav 6.0 an 6.1

Lalbabu2001, 2023-06-26:
> If the ESC calibration is not done properly, the motors do not rotate at the same speeds simultaneously. For this, you need to do the ESC calibration first. You can find lots of videos on Youtube.

I think there is no sense of caliberating esc if the flight controller is sending wrong output to esc,
First we need to ensure the output by flight controller is correct.

Lalbabu2001, 2023-06-26:
> I think i had same problems with 6.0 ill try again

Did you fix it? Please help

paky12, 2023-06-27:
I have not fixed only improved behaviour of motors by tuning PID but still... not enough

paky12, 2023-06-27:
For MPU 9250/6500 do i need logic level converter ? and can i use I-bus instead of S-bus so I avoid signal inverter

rizacelik, 2023-06-27:
This Firmware has been compiled for only 4 engines, if you test and feedback, we will improve accordingly.
[inav_6.1.1_STM32F411CE.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/11879074/inav_6.1.1_STM32F411CE.zip)

Lalbabu2001, 2023-06-27:
I couldn't test this problem on older versions because there is no ppm input,
Sir, if you have mpu6050 please try to test version 6.0 or 6.1 , whether only this doing or all versions have same problem,

paky12, 2023-06-27:
I've tested the firmware 6.1.1 you provided, same problems... When in angle mode one side wont boost the engines when tilted. Could be problems with ESC maybe or i dont really know, controls for controling motors work good. Im still testing some things

Lalbabu2001, 2023-06-27:
Before flashing firmware check the output tab of inav , turn on the precaution slider, and arm then increase throttle, all motor bars should raise, dont connect battery while doing this...

paky12, 2023-06-28:
Hello I just wanted to conclude my solution using MPU6050 I2C, cpu load was not the main problem i think, most of errors were due faulty mixer configuration, low quality gyro and most important PID parameters. I managed to configure everything just enough for drone to "fly", i believe if i used blackbox to record all data and use auto PID to configure correctly it would work perfectly. 
Cheers!

closed 2023-06-28 by paky12

Lalbabu2001, 2023-06-29:
> Hello I just wanted to conclude my solution using MPU6050 I2C, cpu load was not the main problem i think, most of errors were due faulty mixer configuration, low quality gyro and most important PID [...]

Show us your all settings

paky12, 2023-06-29:
I dont have the access to drone anymore, if my colleague has git account hell post here