rizacelik/STM32F411CEU6_INAV_Firmware #16: Motor output isn’t working when setup an airplane

arvet333, 2023-09-19:
INAV 6.1.1
Motor output is not working when setup an airplane while working good when you choose copter.
3,3v on motor pin (A8) when airplane is chosen.

Video of problem: https://youtu.be/E0ATdTfJSdY?si=4-UP1_xoSrIfY0L2

rizacelik, 2023-09-23:
### Note: Firmware update. Please download new firmware.

## Step 1
[image]

## Step 2
[image]

## Step 3
[image]

## ## Step 4. Remove batery

## Step 5
[image]

## Step 6. Connect batery

[image]

# Video

https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/assets/19993109/151d1954-da97-4946-a470-9d4456f63009

## Note: Firmware update. Please download new firmware.

arvet333, 2023-09-23:
Thank you for fixing 🙏

Motors is spinning now, but there is arming flag: ARMING_ DISABLED_PWM_OUTPUT_ERROR

[image]

rizacelik, 2023-09-23:
Hello.
It didn't give me such an error either. Flash the firmware again, if the same error occurs again, I will try to fix it.

arvet333, 2023-09-23:
Reflashed, same error. It will give you this error after all other checks are completed (accelerometer and magnetometer calibrated)

rizacelik, 2023-09-23:
Ok. I did not calibrate. It's my fault. It will be fixed as soon as possible.

arvet333, 2023-09-27:
Seems like I accidentally burned my STM board while testing 😢

Hope all is working fine, but I won't be able to fully test until I get a new board delivered.

PS: We lose 1 servo output after this fix, there is no free compatible pins left to assign?

closed 2023-10-11 by rizacelik