rizacelik/STM32F411CEU6_INAV_Firmware #17: Add MS5611 barometer

stevenferrer, 2023-09-24:
Hi, thanks a lot for your work! I've been trying to connect an MS5611 barometer but it is not being detected. I looked at the `target.h` and it seems [USE_BARO_MS5611](https://github.com/iNavFlight/inav/blob/master/src/main/target/BLACKPILL_F411/target.h#L109) is not defined. Would it be possible to this?

rizacelik, 2023-09-25:
Yes MS5611 barometer. New Firmware and new Connection diagrams have been updated. Please readme file.

stevenferrer, 2023-09-26:
Thanks @rizacelik, tested the barometer with the updated firmware and confirmed that it works 👍  Would really appreciate if you would share the latest code for the firmware. Otherwise, great work and thanks again for sharing the firmware 😄

closed 2023-09-26 by stevenferrer