rizacelik/STM32F411CEU6_INAV_Firmware #18: Multi-color RGB LED strip support option not turn on

Muntaharul, 2023-10-05:
hi the Multi-color RGB LED strip support option is not working how can i turn on this.in my case when i turn on Multi-color RGB LED strip support option then save and reboot the option is still turn off

rizacelik, 2023-10-05:
Hello.
Have you turned on SOFTSERIAL1 RX support in the PORT settings on the control panel? You will use PA8 pin for LED connection.
This address explains in detail how to enable LED STRIP support.
https://github.com/iNavFlight/inav/blob/master/docs/LedStrip.md

Muntaharul, 2023-10-06:
no sir i does not turn on softserial. now  i turn on this option now what need too make change on this option

rizacelik, 2023-10-06:
Test the adjustments on leaf and write the result here. You also help others.

Muntaharul, 2023-10-06:
sir i dont understand leaf mean can you say this clearly adjustment on leaf

rizacelik, 2023-10-06:
Fix LED STRIP. Please download new Firmware and test.

CLI: Enable the LED_STRIP feature via the cli:

**feature LED_STRIP**

Switch to the Configuration tab. There will be a blue tick "Multi-color RGB LED strip support". LED support is enabled.

rizacelik, 2023-10-06:
Firmware has been updated. The error has been fixed. You can test the new firmware by downloading it.
Please fiedback your test result here.

Muntaharul, 2023-10-06:
many many thanks problem is solve thanks 
led is now working Fine

closed 2023-10-11 by rizacelik

suchigno, 2024-05-01:
Hi, in FW 7.1 I have the same problem with the LED strip, if I enable the LED strip in the configuration, save and restart, the option remains disabled. I also tried with the CLI command. Software serials are disabled. I looked at the guide you posted but I can't find anything that explains why there could be a possible conflict. The pin for strip is A8.
Many thanks