rizacelik/STM32F411CEU6_INAV_Firmware #19: PA10 --> GPS issues with DFU mode

zafbangash, 2023-10-06:
The PA10 pin connection with GPS-> TX pin can disable the controller from going into DFU mode in many cases (not every, but many cases).  
I'd suggest if you can switch the gps tx to some other pin, that would help.

zafbangash, 2023-10-06:
Any serial activity on PA10 makes going into DFU difficult for some boards.  For many cases any activity on PA10 (USART_1) makes the bootloader prioritise the pin over the USB, and wont go into DFU mode.  It isn't common issue but it is frequent.

rizacelik, 2023-10-06:
Thank you for the feedback. I tested it and as you said there was a DFU problem. Firmware recompiled and wiring diagrams updated.

zafbangash, 2023-10-07:
Thanks.  Wonderful work

closed 2023-10-07 by zafbangash