rizacelik/STM32F411CEU6_INAV_Firmware #21: GPS not working 6.1

SKVhacks, 2023-10-10:
I'm using ublox Neo 6m GPS module but not working on inav 6.1.1.... It showing red colour GPS logo in app

SKVhacks, 2023-10-11:
> Which pins did you connect the GPS to?

Ublox Neo 6M GPS Rx= A15  ,Tx= B3 and UART 1

rizacelik, 2023-10-11:
I checked and it works for me. 
[image]
Which firmware file did you use?
inav_6.1.1_STM32F411CE.zip
inav_6.1.1_STM32F411CE_HEX.zip
If you used one of these, check your cable connection. It may not be making full contact.

closed 2023-10-11 by rizacelik

SKVhacks, 2023-10-11:
I changed the GPS protocol..... And now it's working..... And I have another dobut is it possible to add buzzer?...

rizacelik, 2023-10-13:
Buzzer not supported.

rizacelik, 2023-10-13:
You will use UART1 for Ublox Neo 6M GPS Module.
Connect PA15 Pin to GPS RX pin and PB3 Pin to GPS TX pin and you need to enable GPS in PORT settings in INAV Configurator 6.1.
PA15 -> GPS RX
PB3 -> GPS TX