rizacelik/STM32F411CEU6_INAV_Firmware #22: Motor 4 doesn't recieve signal after activation of gimbal

Lalbabu2001, 2023-10-24:
As soon as i add  s90 servo gimbal the flight controller cuts signal on m4 pin.

rizacelik, 2023-10-25:
I did not do a gimbal test. Because I've never used it. Unfortunately I don't have it.

Which of the following firmware did you use?
Firmware inav_6.1.1_STM32F411CE.zip OR Firmware inav_6.1.1_STM32F411CE_HEX.zip 

How did you do your installation? Can you give a screenshot? Let me try to help.

Lalbabu2001, 2023-10-27:
I tested all but version 6.1.0 working great.

rizacelik, 2023-10-27:
Did it work on these versions? Thank you for the feedback.
Firmware inav_6.1.1_STM32F411CE.zip VEYA Firmware inav_6.1.1_STM32F411CE_HEX.zip

Lalbabu2001, 2023-10-28:
I didn't tested 6.1.1 because 6.1.0 is fine I am using mpu6500. Is there
anything that activating servo on pin B10 and B1(as per your circuit
diagram) changes the output pin of motor 4 ?

rizacelik, 2023-10-28:
For INAV 6.10 and earlier the connections are different. Please see "Old version INAV Firmware Full wire connection" towards the bottom of the page. There is only 1 servo there. PB1 pin does not belong to servos. Look carefully.
You need INAV version 6.1.1. Links are located at the top of the page. For multiple servos you should use this version.

Lalbabu2001, 2023-10-28:
Even if I use only one pin pb1 for servo it still no signal to motor 4.

Does inav 6.1.1 support ppm?

rizacelik, 2023-10-28:
pin pb1 not servo

[image]

Yes. There is inav 6.1.1 ppm support.

Lalbabu2001, 2023-10-29:
Alright, if i want blackbox log can i connect micro sd card module because
the flash chip is not available here .

rizacelik, 2023-10-30:
Please see diagrams. All the information you need is available there. For Blackbox you will need one M25P16 chip. I'm closing the issue here. Enjoy your work.

closed 2023-10-30 by rizacelik