rizacelik/STM32F411CEU6_INAV_Firmware #23: GPS not working (RED ICON)

ArshaanTech, 2023-10-26:
Hey I am using inav_6.1.1_STM32F411CE_HEX.hex.
The inav configurator shows a red icon for some reason and I think it just isn't working.
Here is the screenshot:
[image]

I have enables GPS in ports at UART1:
[image]

Connections:
```
A15 - RX
B3 - TX
```

rizacelik, 2023-10-26:
Someone asked before and I downloaded the Firmware file and reinstalled it, just like you. I don't see any problems. Here is the video and screenshots.

Check your cable connections. VCC or GND cable may not be making full contact.
Check your Rx and Tx cables, one of them may not be making full contact. Replace if necessary. It should work.

https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/assets/19993109/65bc52f1-d905-4277-b85f-616b2af88bd0

[image]
[image]

rizacelik, 2023-10-26:
[image]

closed 2023-10-30 by rizacelik

ArshaanTech, 2023-11-01:
[image]
Still not working

rizacelik, 2023-11-01:
What is the brand of the GPS module?

ArshaanTech, 2023-11-01:
NEO 6M 0 001

rizacelik, 2023-11-01:
I also have the NEO 6M GPS and tested the firmware GPS function with it. Test your GPS module with a development board such as Arduino. Find out if it works.
It is not a software related problem. I test everything and publish the firmware. If there was a problem, I would get lots of notifications.
Despite everything, I will test the firmware again tomorrow.

ArshaanTech, 2023-11-02:
Thank you so much for the amazing firmware sir! I know that you have tested everything and it must work, I think it might be a problem at my place in Bhopal India. The module is working fine with arduino UNO. I will try more things then. Thank you for the help.

rizacelik, 2023-11-02:
As I promised you, I tested again and this time I installed all the sensors. As you can see on the screen, **GPS** works smoothly.

[image]

ArshaanTech, 2023-11-07:
Okayy thank you so much. I'll just use another module and try maybe that helps. I am still not able to work with it but I'll give u a 1000000/10 for efforts bro. Thank you so much .

shrinibas7, 2024-12-18:
Which uart to select gps after connecting it in b3 and a15

rizacelik, 2024-12-18:
UART1 ->Sensor ->GPS