rizacelik/STM32F411CEU6_INAV_Firmware #24: Standard ESC protocol is not working

arvet333, 2023-10-27:
Reffering to this issue: https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/issues/16

Flashed latest firmware [inav_6.1.1_STM32F411CE.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/blob/main/inav_6.1.1_STM32F411CE.zip) (not HEX)

Have the same problem with ESC: motors activating at 30% throttle and I cannot calibrate it because ESC is beeping on full throttle and not activating.

The problem is reproducible for both airplane and multirotor.

I can record detailed video of problem if needed.

rizacelik, 2023-10-27:
What protocol does ESC use? I have ESC using STANDARD protocol and it passed all the tests.

arvet333, 2023-10-27:
Im using this standard esc: https://www.aliexpress.com/item/1005005446101676.html

They worked fine on 5th INAV version firmware.

Also I tested micro esc - result is the same. https://www.aliexpress.com/item/1005005788101077.html

rizacelik, 2023-10-27:
**In ESC description ESC needs to be calibrated.**

_Safety arming feature: Regardless the throttle stick position, the motor will not spin after battery connected. 
Throttle calibration funtion: Throttle range can be calibrated to be compatible with different transmitters.

Multiple procedure methods supported: transmitter stick, LED procedure Card. Very easy to procedure the ESC at home or at the flying field._

If you cannot calibrate in INAV, my advice to you is to connect your ESC to this pwm output and calibrate it within the maximum and minimum gas range, if your receiver has a pwm output.

The calibration steps are explained as follows in the product description.

**Calibration method:**

_Step 1: Push the throttle lever to its maximum position

Step 2: Plug in the power supply of the electrical regulator

Step 3: Electric adjustment will make two beeps

Step 4: Immediately pull the throttle lever to the lowest position, and the electrical adjustment will emit a beep (3 beeps when connected to 3S, and 6 beeps when connected to 6S) - beep (a long beep at the end indicates successful calibration)_

Try these, if it still doesn't work, I can test it by checking the firmware again.

arvet333, 2023-10-27:
Yep, it doesn’t work. Motor beeping like signal cable isn’t connected on full throttle. So there is no way to calibrate.  When you pull throttle down from 100% to about 50% - then its initialising.

rizacelik, 2023-10-27:
How do you calibrate the ESC? can you make a video?

arvet333, 2023-10-28:
Sure, here it is: https://youtu.be/FB5ciMcrdwo?si=C1_pW30akq6aij4T

rizacelik, 2023-10-28:
I watched your video. As you said, the throttle does not detect the desired signal level when it is at 2000. The value 2000 appears to be above the ESC signal range. It is not within the range required by ESC. ESC is not fully calibrated. It seems so. Have you tried other protocols? It may not support the STANDARD protocol.

Have you tested on other pins? B5, B6, B7.
Do you have the same problem with these pins?

The brand of the ESC product you are using is not written on it and no information is given about the protocols. If you know the brand, you can find information about this problem by searching the internet.

If the problem still persists, you can ask about it by opening a topic on the RC forums and maybe find a reasonable answer.

I am not making any changes to the ESC protocols. I use whatever the original codes are.

There was someone reporting a similar ESC problem before, but the problem was fixed. There were no other notifications about this and I had no problems. After doing all the tests, I share the files.

rizacelik, 2023-10-28:
**NOTE:** Your ESC sees the maximum gas setting as 1100. This indicates that the product is still in protection mode as described in the software.

**Safety arming feature: Regardless of the throttle stick position, the motor will not spin after the battery is connected.**

rizacelik, 2023-10-28:
Bu videoda görüldüğü gibi PWM çıkışı olan bir RC alıcı ile ESC kalibre edilebilir. Bunu deneyin. Sorununuz çözülecektir.
https://youtu.be/yaAf4eofppc

arvet333, 2023-10-28:
That problem was fixed by you in september, but it apeared again with newer versions of firmware. I have an old version of firmware on my computer, flashed it and all working fine, ESC was calibrated without any issues.

Here is the video: https://youtu.be/Xm1HkSAMQRc?si=e2AA1G7ca3HMGfd5

I guess my "old version" is somwhere on red line in a history
[image]

rizacelik, 2023-10-28:
If it works on older versions then there is a TIMER conflict.
I want to know: Is there a similar problem on B5, B6, B7 pins?

Please test them all. This is necessary to find the TIMER Conflict.

arvet333, 2023-10-28:
hm. after reflashing seems like it work fine. need some time to test it properly

arvet333, 2023-10-29:
Seems like everything is ok, so Im closing it

closed 2023-10-29 by arvet333

rizacelik, 2023-10-30:
It's nice that everything went well. Good luck.