rizacelik/STM32F411CEU6_INAV_Firmware #25: Seems like voltage isn't working properly

arvet333, 2023-11-01:
VBAT voltage isnt changing.

Here is detailed video of testing: https://youtu.be/y0L-hrXLQsw

rizacelik, 2023-11-01:
You cannot measure voltage and current directly. For this, you must use a Current Sensor ACS712. When you connect a 2S or 3S battery directly to the pin, the Board lights up.
Using ACS712 sensor with Arduino: https://www.youtube.com/watch?v=1S7vQubTgQg

arvet333, 2023-11-02:
How we can measure **voltage** with **current** sensor?

**For measuring current:**
Use Current Sensor ACS712. Connect ACS712 OUT pin to FC B1 pin.

**For measuring voltage:**
Use voltage divider scheme with resistors. Maximum battery voltage after dividing should be below 5v. Connect battery plus (after dividing, max 5v) to B0 pin on FC. Battery GND to FC GND.

Am I right?

In my case there is 1S battery, so it is below 5v (4,2v maximum), so voltage divider is not needed, But as you can see in the video: INAV voltage indicator is stuck and not changing when real battery voltage is changing.

rizacelik, 2023-11-02:
Yes, you are right. You can search by changing the Voltage Scale value.
[image]
You also need to change the battery information.
[image]

arvet333, 2023-11-02:
So now I can say for sure: voltage measurement is not working.

As you can see from my video: voltage scale number is set correctly, battery information is also set correctly.

**Testing algorithm:**
1. Measuring real battery voltage -  3,7v in my case
2. Finding right scale value to get 3,7v in INAV
3. Charging battery to 4,2v and connecting to FC
4. INAV is **still showing 3,7v instead of 4,2**

rizacelik, 2023-11-02:
I've never tested it. If I have time, I will review this by looking at the source code. If there is a problem, we solve it together.

rizacelik, 2023-11-02:
The code that calculates the ADC battery level is given as follows.

```
#ifdef USE_ADC
uint16_t getVBatSample(void) {
    // calculate battery voltage based on ADC reading

    // result is Vbatt in 0.01V steps. 3.3V = ADC Vref, 0xFFF = 12bit adc, 1100 = 11:1 voltage divider (10k:1k)

    return (uint64_t)adcGetChannel(ADC_BATTERY) * batteryMetersConfig()->voltage.scale * ADCVREF / (0xFFF * 1000);
}
#endif
```

Voltage Scale meaning : 1100 = 11:1 voltage divider (10k:1k)

arvet333, 2023-11-02:
There is 3.3v figurating in this code. May be 3.3v is the maximum, not 5v. Let me do some testing with it.

arvet333, 2023-11-02:
Yep, if the voltage is below 3.3v all is working well. I sent a PR to add this information to readme file, closing issue now.

rizacelik, 2023-11-02:
You did a good job. If everything is fine with 3.3V, the calculation will be as follows.

[image]