rizacelik/STM32F411CEU6_INAV_Firmware #27: fc not working

ArshaanTech, 2023-11-18:
Hey bro, My flight controller suddenly stopped working.
I have flashed firmware 3 times, it flashes and connects to iNAV but the sensors are not working :(.

ArshaanTech, 2023-11-18:
I also noticed that the Gyro keeps blinking anonymously. it flashes stops randomly. The connections are correct and I have checked the gyro with ESP32, it is working.

rizacelik, 2023-11-18:
There is a possibility. Your cables are not making full contact with the sensor. That's why it might not be working. Check the cables. If you soldered, re-solder. Sometimes cold solder can cause contact problems. Can you share a screenshot of your links?
Can you share the INAV flight controller SETUP screen? I want to see.

ArshaanTech, 2023-11-18:
you were right, I did the soldering again and it works :). Thanks

ArshaanTech, 2023-11-18:
By the way I wanted to know if you will work for making any firmware on betaflight... any plans?

rizacelik, 2023-11-18:
I'm glad you solved the problem. I can make a firmware for Betaflight, but my time is very limited for now. Maybe later.

ArshaanTech, 2023-11-19:
Okay bro, if possible whenever you have time just make a firmware for betaflight too (STM32f411ceu6). 
It would be really great! Can I make a video of your projects on my YouTube channel?

ArshaanTech, 2023-11-19:
There's another problem now, I calibrated everything and everything is working fine but the board isn't powering up from 5V.
Which is quite strange because the sensor and receiver is powering up. What could be the issue?
[image]

rizacelik, 2023-11-19:
Wrong connection wire.
[image]

Use by making multiple connections from one cable.
[image]

It must be a 5V BEC source. So you can use the 5V you get from the ESC. Its current should not be low. 5V should not be low, it should be full 5V. It may not have opened because the current was not enough. A value above 5V will burn the card.

ArshaanTech, 2023-11-19:
okay sir, i will try that tomorrow. Thank you.
The BEC output is 5v 3A. I think it should be enough. I checked on the multimeter, it was 5.21 volts.

ArshaanTech, 2023-11-20:
still not powering up.

ArshaanTech, 2023-11-22:
What should I do?

rizacelik, 2023-11-22:
I think something happened to the 5v input you are using. Have you tried the 5v here?
[image]

rizacelik, 2023-11-22:
Also is it ARM enabled? If ARM is not enabled, it still won't open.

venkatkamesh, 2023-11-23:
> Also is it ARM enabled? If ARM is not enabled, it still won't open.

can you please share your mail id .
my mail id is venkat.kamesh90@gmail.com
Thank you waiting for your reply

rizacelik, 2023-11-23:
### Controls, Arming
Perform calibration operations. Everything should be a green tick on the right side of the setup screen.
The arming process is the final stage where your Quadcopter or other vehicle is ready and approved to fly. Your plane will not fly without the Armin process. To do this, you need to open the Armin process from the inav-configurator screen and the motors and servo outputs from the Output tab.
[image]

[image]

Bunları yap

rizacelik, 2023-11-23:
If you are not familiar with ARM, search the YouTube channel. You can find lots of videos. I hope I could help.

ArshaanTech, 2023-11-23:
the arm is working on USB, I'll try different 5V and ground.
Btw I found https://github.com/betaflight/unified-targets/blob/master/configs/default/FLWO-FLYWOOF411V2.config
any idea how do I make a betaflight firmware out of it? if you can help me out

venkatkamesh, 2023-11-23:
> If you are not familiar with ARM, search the YouTube channel. You can find lots of videos. I hope I could help.

had problem with the mpu6050 on by compilation . any suggestions ? (working on latest 7 source)
Need help friend . since mpu6050 is removed from latest source and i made a revert of implementing the mpu6050 still not working.
any suggestions or help is really appreciated . 
Thank you

rizacelik, 2023-11-24:
I am only publishing Firmware here. I have detailed how to make the connections in the README file. I don't know why such problems occur. Make sure you use and connect the correct Pins. If it still doesn't work, you may have probably burned the module.
Like I said, I can only tell you the possibilities. I can't explain why it doesn't open.

venkatkamesh, 2023-11-26:
> I am only publishing Firmware here. I have detailed how to make the connections in the README file. I don't know why such problems occur. Make sure you use and connect the correct Pins. If it still [...]

since open source and based on your amazing stuff , i am curious to know the changes in source.
My purpose is not to make a firmware to competition . its for my personal with osd balckbox with SD enabled

ArshaanTech, 2023-11-26:
Hey bro I just needed a help. Is it possible for you to send me the configuration CLI file of the firmware? like the setting for sensors i2c, SPI motor outputs Softserial etc. i just need that .

rizacelik, 2023-11-26:
There is nothing to compete with. It is very difficult for me to provide you with hardware related assistance. Because if you connected the sensors incorrectly, there is a possibility that you may have burned them.
If you connect your sensors correctly according to the connection diagram, you will not have any problems. Therefore, check the sensor pins at least 3 times when soldering or connecting them to the board. You should be very careful, especially for GND and VCC inputs. Some sensors work with 3.3V, if you accidentally connect 5V your sensor will burn.
You do not need your CLI configuration information. Once the firmware is installed everything will work fine.
You will only need the following CLI command. MPU6050 Module is very slow. That's why the Microcontroller load goes up to 90%. To avoid this, execute the following CLI command.

set looptime = 1500

For performance, you can use the MPU6500 module with SPI2

venkatkamesh, 2023-11-27:
> There is nothing to compete with. It is very difficult for me to provide you with hardware related assistance. Because if you connected the sensors incorrectly, there is a possibility that you may [...]

Thank you for the Tips my friend 
wiring and all other things were so perfect
the clock speed isn't matching the mpu6050 i2c

When i changed the below clock speed in system_stm32f4xx.c as below

```
#if defined(STM32F410xx) || defined(STM32F411xE)
#define PLL_N      336
/* SYSCLK = PLL_VCO / PLL_P */
#define PLL_P      4
/* USB OTG FS, SDIO and RNG Clock =  PLL_VCO / PLLQ */
#define PLL_Q      7
```

then my blackpill started to work with usb communication with 25 HSE Value
all other sensors with i2c are working except mpu6050. i believe you cracked it and i request for suggestions
Changing looptime isnt working. 
* when i restart boot isnt happening until i remove mpu6050,. after boot when i connected it back it does no init
* uploading your latest ones 6.1.1 the mpu6050 is working marvelously.

please do share your suggestion thank you

rizacelik, 2023-11-27:
INAV supports STM32F411CEU6 board in the latest version. You can look from here.
https://github.com/iNavFlight/inav/tree/master/src/main/target/BLACKPILL_F411

I did not make any additional settings for the MPU6050. INAV no longer uses the MPU6050 sensor. Because it is very slow and increases the CPU load by 70%. I won't use it in the next compilation either.

rizacelik, 2023-11-27:
You can use MPU9250. It uses MPU6500. You can use Gyroscope and Accelerometer support with SPI. You can also get MAG and BARO support via I2C.
Thank you.

ArshaanTech, 2023-11-27:
bro i just need a configuration file,
it looks smth like this: 
```
set gyro_hardware_lpf = NORMAL
set gyro_lpf1_type = PT1
set gyro_lpf1_static_hz = 250
set gyro_lpf2_type = PT1
set gyro_lpf2_static_hz = 500
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set gyro_calib_duration = 125
set gyro_calib_noise_limit = 48
set gyro_offset_yaw = 0
set gyro_overflow_detect = ALL
set yaw_spin_recovery = AUTO
set yaw_spin_threshold = 1950
set gyro_to_use = FIRST
set dyn_notch_count = 3
set dyn_notch_q = 300
set dyn_notch_min_hz = 150
set dyn_notch_max_hz = 600
set gyro_lpf1_dyn_min_hz = 250
set gyro_lpf1_dyn_max_hz = 500
set gyro_lpf1_dyn_expo = 5
set gyro_filter_debug_axis = ROLL
```
AND
```
resource MOTOR 1 A08
resource MOTOR 2 A09
resource MOTOR 3 A10
resource MOTOR 4 B00
resource MOTOR 5 B04
```

So could you send me the settings like these u have changed in the firmware? please?

rizacelik, 2023-11-28:
I don't have a ready-made card. The card used after the tests is used for other purposes. That's why I can't give you a CLI output with all settings. Not every CLI output may be useful to you. It won't do you any good to ask for something you can do in just a few simple steps.

1. Connect or solder your modules to the board as shown in the diagram.
2. Load the firmware.
3. Calibrate the modules (sensors) gyro and accelerometer sensor.
4. Calibrate the ESC.
5. Set your transmitter's protocol. SBUS,iBUS, PPM etc.
6.Turn on the engine output.
7. Define the ARM function from the mode selection.

This will be what you did in the CLI records. Other steps include ESC connection of motors, battery connection etc.

NOTE: I don't have much time and I have other things to do. Unfortunately, I won't be able to write to you from now on. Thank you for your understanding.

ArshaanTech, 2023-11-28:
Hey for some reason my MPU6500 is not working :(
[image]
it shows up gyro and accelerometer but values aren't changing.

closed 2023-11-28 by rizacelik

venkatkamesh, 2023-12-03:
> Hey for some reason my MPU6500 is not working :( [...]

hey man,
check if both osd / blackbox are enabled with same cs pin of mpu6500 . in such case this could happen
or just comment the spi1 and osd and blackbox. done there you go. mpu6500 works