rizacelik/STM32F411CEU6_INAV_Firmware #29: Blackbox

Lalbabu2001, 2023-12-24:
Where to connect blackbox connection Quadcopter configuration . Can I use SD card module https://images.app.goo.gl/5wFvNcc6KCMUzVYJ6

rizacelik, 2023-12-24:
Unfortunately, there is no SD card support. You can use **M25P16** instead. Here **M25Pxx** represents the **xx** flash size. You can get it in any size you want.
Connection pins are as follows.

SPI2_SCK_PIN           **PB13**
SPI2_MISO_PIN         **PB14**
SPI2_MOSI_PIN         **PB15**
SPI2_CS_PIN              **PA4**

If you can't find this flash chip in your country, you can buy it from Aliexpress.
https://aliexpress.com/item/1005003030545704.html

Lalbabu2001, 2023-12-25:
> Unfortunately, there is no SD card support. You can use **M25P16** instead. Here **M25Pxx** represents the **xx** flash size. You can get it in any size you want. Connection pins are as follows. [...]

Can I use w25q32 flash module, it's easily available here. Do I also need txs0108e module?

Lalbabu2001, 2023-12-26:
Kindly assist me

rizacelik, 2023-12-26:
It probably supports the w25q32 chip, but I haven't tested it. It should work normally.

Lalbabu2001, 2023-12-26:
Do I need txs0108e module or I can do without it?

rizacelik, 2023-12-26:
Check the pin level of the module. The datasheet file contains the necessary information. If it is 3.3V, you will need a level converter. If it is 5V there is no need.

rizacelik, 2023-12-26:
No, it's okay. Because you can connect multiple modules to the SPI interface. Because the CS pin of each connected module is different.
SPI2_CS_PIN PA4
This pin is reserved for M25P16.

Example
[image]

Lalbabu2001, 2023-12-27:
Right now you are serving us with knowledge, in future we will serve new generation with knowledge provided by you. Thank you for your time sir.

closed 2023-12-27 by Lalbabu2001