rizacelik/STM32F411CEU6_INAV_Firmware #31: Battery voltage error, and led strip error in inav 7.0 firmware

Karan1rao, 2023-12-27:
In STM 32 f411ceu6, I am not able to measure voltage accurately, I have properly divided voltage to get voltage of around 5 volts, however when I connect my battery to the flight controller it shows some voltage which can be fixed by adjusting voltage scale, however, when another battery of same cell count with higher or lower voltage is connected inav does not show the correct voltage, furthermore, i guess the led strip function is not working in inav 7.0 firmware

rizacelik, 2023-12-27:
You need to make your calculation as 3.3V. Calculations are given below. 5V does not give the actual value.

Did LED INAV 6.1.1 also work? If it works, I can look for INAV 7.0.

[image]

rizacelik, 2023-12-27:
NOTE: Voltage pin will be PB0 in STM32F411CEU6. The above diagram is for STM32F405 board.

Karan1rao, 2023-12-27:
Yes the led works in 6.1.1 firmware. And I would like to ask one more question, is the buzzer enabled in inav 7.0 firmware?

rizacelik, 2023-12-27:
buzzer not supported.

Karan1rao, 2023-12-27:
Voltage divider for 3.3 volts worked for me, thanks.       
Led is also working 🙂🙂, thanks bro.

rizacelik, 2023-12-27:
It's great that everything went well. I wish you good work. Enjoyable flights.

Karan1rao, 2024-01-01:
Bro how do i connect a telemetry Tx and Rx pin to the board as one Tx and Rx pin is used by  GPS and one Rx pin is used as PPM input, furthermore, i need to connect an osd to the controller and spi port is used by MPU 9250, do i connect osd with the same spi pin as of mpu 9250?