rizacelik/STM32F411CEU6_INAV_Firmware #32: Betaflight 4.5.0 motors not working.

ArshaanTech, 2023-12-28:
Motor outputs are not working...

[image]

rizacelik, 2023-12-29:
OK, I'll look later. Thank you for the notification.

rizacelik, 2023-12-29:
If you use MPU6050, ARM operation will not occur. You should use MPU9250 or MPU6500. Calibrate these. If you use compasses, calibrate them.
Do the ARM transaction. You cannot turn on the motor output without ARM operation. After activating the motor output, the motors will start to rotate.

Please do these. If the engines do not start after doing these, report them.
I tested it and it works.

rizacelik, 2023-12-29:
There is also one more detail. Betaflight engine outputs do not work when the battery is not installed. You need to connect the battery while the board is plugged into USB. Be careful not to burn the Borad.
Watch this video.

https://youtu.be/1WYDsiYJGoQ?t=796

ArshaanTech, 2023-12-29:
it worked as soon as i connected the MPU6500 sensor. Thank you.

closed 2023-12-29 by ArshaanTech