rizacelik/STM32F411CEU6_INAV_Firmware #33: MPU6050 is working when we build INAV 7.0.0

NilsLab, 2023-12-31:
The addition of MPU6050 support in INAV 6.1.1 and INAV 7.0.0 is excellent work ,  I have added MPU6050 to INAV 6.1.1 but the board is not booting after changes in common_hardware.c, I might be missing some changes since you added MPU6050 in INAV6.1.0 and INAV 7.0.0, can you help hear?

```
    #if defined(USE_IMU_MPU6050)
        BUSDEV_REGISTER_I2C(busdev_mpu6050,     DEVHW_MPU6050,      MPU6050_I2C_BUS,    MPU_ADDRESS,        GYRO_INT_EXTI,  DEVFLAGS_NONE,  IMU_MPU6050_ALIGN);
    #endif
```

NTH106, 2023-12-31:
After I loaded the hex file (inav_7.0.0_STM32F411CE.rar) into the circuit and tested it, it failed to connect and displayed 3 lines like this "

-- MSP connection successfully opened with ID: 1
-- No configuration received within 10 seconds, communication failed
-- MSP connection successfully closed"

rizacelik, 2023-12-31:
@NilsLab 
No, nothing changes.

@NTH106 
Have you uploaded the hex file and connected the sensors if it doesn't work? Which sensors did you connect? There is no explanation. You'll appreciate that I can't understand what the problem is just by saying it doesn't work.
State the problem more clearly and clearly. Write what you used.

NilsLab, 2024-01-01:
@rizacelik
INAV 6.1.1 and INAV 7.0.0 Firmware from your repo works fine, There is no issue.

I was building my firmware with MPU6050 changes that are not working, I have backported MPU6050 changes to INAV 6.1.1 and INAV 7.0.0, But the board is not getting connected to the configurator.

The issue may be with the backporting of MPU6050 or some changes needed in the MPU6050 driver or somewhere else to work on INAV7.0.0, Since you have done the same work for your firmware build you can help me.

If you're ok with sharing a patch of MPU6050 to support INAV7.0.0 that will be a great help. 

I want to have some pin changes, I am trying to make one PCB board blackpill but before I need to see if the firmware works with pin changes, or not.

NilsLab, 2024-01-01:
With MPU6050 inav 5.1.0 not booting
I am facing an issue with the MPU6050 build with Inav 5.1.0
only the following changes are done for 5.1.0
1. target.c and target.h used from your repo
2. `target_stm32f411xe(STM32F411CE HSE_MHZ 25)`
3. updated src/main/target/system_stm32f4xx.c
```
  #if HSE_VALUE == 24000000
+    #if HSE_VALUE == 25000000
+        #define PLL_M   25
+    #elif HSE_VALUE == 24000000
         #define PLL_M   24
     #elif HSE_VALUE == 16000000
         #define PLL_M   16
```
If I removed the MPU6050 define from the target.h board is booting fine
#define USE_IMU_MPU6050
#define IMU_MPU6050_ALIGN       CW0_DEG
#define MPU6050_I2C_BUS         BUS_I2C1

Can you suggest what might be missing on inav5.1.0 to get MPU6050 working

rizacelik, 2024-01-01:
I follow the steps here. I didn't make any other changes.

https://github.com/iNavFlight/inav/commit/1c536ccb837e5e06e2936a69d99de8636afff758

Removed support information is in red. Place them in the relevant files.

NilsLab, 2024-01-01:
I have retried the above changes but still facing the same issue , Board is not booting with MP6050
Can you please provide diff with your local repo  this will help me to go ahead
`$ git diff HEAD > mpu6050.patch `

I have attached my `$ git diff HEAD > mpu6050.patch ` to this comment if possible have a look at this one
https://github.com/NilsLab/INAV_7.0.0_BlackPill/blob/main/INAV_7.0.0_STM32F411_MPU6050.patch

rizacelik, 2024-01-01:
No, there is nothing different. Just insert the removed codes for INAV support. There are no other changes.

Why is MPU6050 so important? This sensor is not suitable for flight. It increases the CPU load by 80%. Additionally, INAV announced that it will no longer support the STM32F411 board.
MPU6050 sensor Betaflight does not work at all. The card does not do ARM. I think you should not waste your time.

NilsLab, 2024-01-02:
I had a bunch of MPU6050 with me and it was very cheap 
It would be great if you could help me get going.
But there is a chance of some change in the source code that you have, because inav5.1.0 is not working with MPU6050 when built with provided target files, your repo all FW works fine with MPU6050.

Lalbabu2001, 2024-01-16:
Yeah for just for few bucks you should not risk into instability.

NilsLab, 2024-01-17:
I have moved my DIY build with MPU6500 and it working fine 
We should not use local changes on the latest release,
Now I have a clean INAV 7.0.0 build which only has a new Board definition as my connection, no source code change

NilsLab, 2024-01-17:
Closing this issue since is's not supported officially on INAV 7.0.0

closed 2024-01-17 by rizacelik

Lalbabu2001, 2024-01-18:
What's your parts specifications?