rizacelik/STM32F411CEU6_INAV_Firmware #35: Blackbox

Lalbabu2001, 2024-01-01:
Hello again sir, hope you are doing well.
I connected w25q32 module as per your suggestion.
I also selected spi option in configurator.
But it's not detecting
I am powering stm32 with 3.3v so directly connected the flash module to stm32.
Inav blackbox page shows it has support for w25qxx series chips.
[image]

rizacelik, 2024-01-01:
There is W25Qxx support for INAV. You can read it here. You need to turn on support via CLI command. Please read.

https://github.com/iNavFlight/inav/blob/a90203cac97efb9c6a9879002a1371c32ccc250a/docs/Blackbox.md#onboard-dataflash-storage

Lalbabu2001, 2024-01-02:
I turned on but not working. It's not detecting.
I'm sharing my wiring diagram please verify it😊
[image]

Lalbabu2001, 2024-01-03:
Only gyro mpu6500 is connected spi as per diagram.

Lalbabu2001, 2024-01-16:
Is there something wrong with hex?

rizacelik, 2024-01-16:
No there's not. How did you check if W25Qxx is working? According to INAV documentation, this chip should be working.

Lalbabu2001, 2024-01-17:
I checked as follows: wire everything as per image I posted earlier.
1.I just turn on blackbox feature in configuration tab
2. In cli I set it to SPIFLASH
3. Rebooted but the configurator says no chip found besides the connect button.
4. Is there something left?

rizacelik, 2024-01-17:
If I can find the W25Qxx chip I will test it and let you know. You have already covered the technical details regarding its use. If it doesn't show up in the configurator, it looks like it's probably not supported.
There's nothing I can do about this.

Lalbabu2001, 2024-01-17:
It's ok sir 😊, We know you help us to the extent you can, respect from India 🙏🙏

longlong7211, 2024-01-17:
I have a W25QZZ 16MB chip, I just need to connect the wires and it works fine. I have tested it on INAV version 6.0.0, and although I haven't tried version 7.0.0, I am confident that no additional configuration is needed.

rizacelik, 2024-01-17:
@longlong7211 It's nice that it works. Thank you for your help. I hadn't tested it and had no idea about it. I hope @Lalbabu2001 manages to make it work too.

longlong7211, 2024-01-17:
I think it's a common issue with the INAV software. Perhaps you should reflash the firmware. I've encountered cases where the GPS module had a solid blue icon but couldn't achieve a 3D fix, even the PPS light on the module wasn't blinking. However, after reflashing the firmware, it started working. This issue also occurred with the gyro, not just on this board but on others as well. I found the most accurate solution is to solder all the sensors first and then flash the firmware; it tends to work better and reduces the chances of errors.

rizacelik, 2024-01-17:
Absolutely I agree. While soldering the connections, pin connections should be checked at least 3 times. In case of cold soldering, the soldering iron should be well heated and quality solder and flux should be used. It may cause contact problems in cold soldering.
Sometimes people can overlook it and cause incorrect soldering of pins.

Lalbabu2001, 2024-01-18:
Hola sir it's now working I used another module w25q128. You are great 👍

rizacelik, 2024-01-18:
I congratulate you. You did it.

mursa1en, 2024-04-17:
Hello, Mr Celik 
Hope you're doing great. I have a question: is it possible to use this SPI SD card adapter for black box. Also I have recently tried to contact the vl53l1x and vl53l0x range finder over i2c but unfortunately inav didn't detect both of them. I'm still using the inav 7.0. Can you please help me with this?[image]

rizacelik, 2024-04-17:
Not support SD card.

Lalbabu2001, 2024-04-20:
Bro just buy a wq2864 module

Harshhhh16, 2024-05-04:
Hello as your'e suggestion i used inav 7.1 but there is some dought
1.  we need to add logic level convertor for gyroscope ? i used bmi160 so whats the connection it should be connected to the LLC and to the board or direct to the board ??
2.  where we connect the black box can you explain pin i don't understand where to connect

rizacelik, 2024-05-04:
jiroskop INAV 6.1.1  ne ise INAV7.1 de de aynıdır. 
INAV 7.1 bağlantıları README dosyasına verilmiş. Kara kutu için MSP16 olarak verilmiş pin bağlantısı var.  MSP16 sizde yoksa yerine wq2864 kullanabilirsiniz. Pinler aynıdır.

Panele bağlantıyı INAV 6.1 de nasıl yaptıysanız aynısı.
Bütün bilgiler README dosyasında mevcut.

rizacelik, 2024-05-04:
The gyroscope is not the same as INAV7.1 is what INAV 6.1.1 is.
Links to INAV 7.1 are provided in the README file. There is a pin connection given as MSP16 for the black box. You can use MSP16 alongside or instead of wq2864. Pins are the same.

The panel layout is the same as in INAV 6.1.
All information is available in the README file.

rizacelik, 2024-05-04:
For each INAV version, you must download the configuration for that version. In short, for the INAV 6.1.1 version, you need to download and use the INAV 6.1.1 configuration.
For INAV 7.1 version, you need to download and use INAV 7.1 configuration.

Harshhhh16, 2024-05-05:
thank you so much it's help me a lot 
and according to diagram in spi connection i connect bmi180 and bmp280 together & in i2c connection i connect qmc5883l & bmp 180  together it's correct ?? and thank you :)

rizacelik, 2024-05-05:
bmi180 ??? 
bmp280 ve bmp180 ikisi de barometre birisini kullanacaksınız. Lütfen biraz araştırma yapın. Modüller hakkında bilginiz olsun.

Harshhhh16, 2024-05-07:
ohh sorry now i understand if i use i2c connection then bmp180 is used and if i want to use spi connection then bmp280 used :)