rizacelik/STM32F411CEU6_INAV_Firmware #39: Telemetry

Lalbabu2001, 2024-01-16:
I wanted to know does this flight controller support telemetry output?

rizacelik, 2024-01-16:
You can use SOFTSERIAL for telemetry. You can turn it on from PORT settings. You may also need to configure it via CLI. You can learn the details by looking at INAV documents.
The pins you need are defined as follows. This is valid for quadquadcopter.

```
SOFTSERIAL TX PIN     PA0
SOFTSERIAL RX PIN     PA1
```

Harshhhh16, 2024-05-03:
hello i used 6.1.1inav version but where to connect telemetry ?? i don't find serial port can you please tell me where to connect ??

rizacelik, 2024-05-03:
What is your telemetry device?
The serial connection in INAV6.1.1 is as follows.

```
SOFTSERIAL TX PIN     PA0
SOFTSERIAL RX PIN     PA1

```

Why don't you switch to INAV 7.0 or INAV 7.1? you should do this. To benefit from the improvements, you must upgrade to INAV 7.1.

Support INAV 7.1 Serial

```
SOFTSERIAL TX PIN     PA9   // Softserial1_TX
SOFTSERIAL RX PIN     PA10  // Softserial1_RX

SOFTSERIAL TX PIN     PA0  // Softserial2_TX
SOFTSERIAL RX PIN     PA1 // Softserial2_RX
```

Harshhhh16, 2024-05-04:
as your'e recommendation i switch to latest version but which is better accelerometer ? BMI160 vs MPU9250 vs MPU6500 ??

rizacelik, 2024-05-04:
they are all good

Harshhhh16, 2024-05-16:
hello i connect properly A0 And A1 pin to telemetry TX & RX but there is no data receive can you solve the problem ???

[image]

is this correct ??

rizacelik, 2024-05-16:
The answer I wrote to you is above. There is the following information there. It says which pins to use for which serial.

```
SOFTSERIAL TX PIN     PA9   // Softserial1_TX
SOFTSERIAL RX PIN     PA10  // Softserial1_RX

SOFTSERIAL TX PIN     PA0  // Softserial2_TX
SOFTSERIAL RX PIN     PA1 // Softserial2_RX
```

After testing, write the result here. Others will also benefit from this information and feedback.

Harshhhh16, 2024-05-16:
All connection is same as you mentioned above 
I connect gps to PA9 PA10 its work
But Telemetry in PA0 PA1 is not working 
UART 1 is for gps 
UART 2 is for sbus
Soft serial 1 is for telemetry 
I’m right?

rizacelik, 2024-05-16:
Have you looked at the diagram? Which pins is GPS connected to?
Use the following pins for telemetry.

```
SOFTSERIAL TX PIN PA9 // Softserial1_TX
SOFTSERIAL RX PIN PA10 // Softserial1_RX
```

If these pins do not work then install inav_7.1.0_STM32F411CE.zip.
Use the pins below when you upload this file.

```
SOFTSERIAL TX PIN PA0 // Softserial2_TX
SOFTSERIAL RX PIN PA1 // Softserial2_RX
```

Harshhhh16, 2024-05-16:
[image]

I prefer this connection is it fine or I want to change it ?

Harshhhh16, 2024-05-16:
Hello Now It's Work Properly i connect telemetry to PA9 & PA10
but in telemetry output is show 25v battery or 34v battery my battery is 3s 2200mah and also i connect resistor that is i didn't get 62k resistor so i connect 51k+10k+1k in series and 22k resistor still it's not measure correct voltage what's the problem ??

rizacelik, 2024-05-16:
Which telemetry module are you using?

Harshhhh16, 2024-05-16:
I saw a video on youtube that he makes hc12 module telemetry and alternate way is by hc05 module he also build with stm32 f411 from your github page his telemetry working fine but i think in my telemetry also  all parameters working its show altitude,navigation fine buy only battery voltage is incorrect…!

rizacelik, 2024-05-16:
There will be a 62k resistor on the positive side of the battery. Watch out for this.

rizacelik, 2024-05-16:
You can try the calculation below.

[image]

Harshhhh16, 2024-05-16:
Yess i follow diagram 62k on n and 22k on negative terminal of battery and rest terminal common and goes to port B0 is I need to change 62k resistor to 14k ?

rizacelik, 2024-05-16:
No. Use the red colored resistor value given in the diagram above.
Do not connect incorrectly. Board may burn.

rizacelik, 2024-05-16:
[image]

Harshhhh16, 2024-05-16:
Ok i go with this diagram let’s see if it’s work

rizacelik, 2024-05-17:
Were you able to do it?

Harshhhh16, 2024-05-17:
i recently burn out the board may be for wrong connection and also vl53lox sonar sensor also detected but it's need any calibration or set any mode ?? or it just plug and play
and second is how to setup ws2818 led strip ?? i don't understand it
rest all are working fine that's very nice and also thanks to you

AaronSmith69, 2024-08-12:
> I saw a video on youtube that he makes hc12 module telemetry and alternate way is by hc05 module he also build with stm32 f411 from your github page his telemetry working fine but i think in my [...]

do you have a video of that? i tried to build telemetry with hc-12 but it didn't get any info back to my GCS.