rizacelik/STM32F411CEU6_INAV_Firmware #41: inav 7 doesn't work

olegin62, 2024-02-16:
The configurator does not connect to the controller after flashing the firmware. After downloading, LED C13 does not blink for one version and is permanently lit for the second version.

rizacelik, 2024-02-16:
It should work. Reinstall the firmware. Check your pins. In case of lack of contact, problems may arise. Re-solder your solders.

olegin62, 2024-02-16:
Version 6 works fine. Is it possible that 6 works and 7 doesn't?

rizacelik, 2024-02-16:
Since no changes were made from INAV 6.1 to INAV 7.0, it should work and has been tested.

For INAV 7.0 you need to install and connect with INAV Configurator 7. Have you done this?

Select "FULL CHIP ERASE" when installing.

Download and install the new [INAV Configurator 7](https://github.com/iNavFlight/inav-configurator/releases)

olegin62, 2024-02-16:
I checked on 7.0.0 and 7.0.1 versions of the configurator, same result, after flashing the firmware the LED flashes 3 times then 4-5 fast flashes and that is it.

rizacelik, 2024-02-16:
Along with the board, what sensors are you using?

olegin62, 2024-02-16:
Currently I don't use any sensors, there is only board

Dasmen, 2024-02-17:
Возможно у Вас в настройках включены osd или sdcard, но физически они не подключены. У меня так было. Отключил в таргете, собрал прошивку и все заработало.

rizacelik, 2024-02-17:
[olegin62](https://github.com/olegin62) It may not work without connecting a sensor. New versions may require sensor configuration. Please use the sensors specified in the diagram. It will work.

Dasmen, 2024-02-17:
> [olegin62](https://github.com/olegin62) It may not work without connecting a sensor. New versions may require sensor configuration. Please use the sensors specified in the diagram. It will work.

Please build the firmware from without FEATURE_OSD to outer, I think it doesn't work because of this.

olegin62, 2024-02-20:
> [olegin62](https://github.com/olegin62) It may not work without connecting a sensor. New versions may require sensor configuration. Please use the sensors specified in the diagram. It will work.

Thanks, it fixed the problem.

closed 2024-02-20 by olegin62