rizacelik/STM32F411CEU6_INAV_Firmware #42: Issue with the mpu9250

mursa1en, 2024-02-18:
[image]

Hello sir.
I'm using the MPU9250 and have connected it over SPI but inav detects  it as the mpu6500. I have tried to select the MPU 9250 manually but it didn't work. Now i have to use the HMC5883. Can you please help me with this issue.

rizacelik, 2024-02-18:
If INAV detects it as MPU6500, it is correct. You bought a fake module.
You need to connect the HMC5883 compass as given in the diagram. This is given very clearly.

BowenL1, 2025-02-27:
Hi Sir

I have the same problem. but I can't detect any gyro accel and mag sensors even I used MPU-9250/6500 (MP92 421LC1 2426 on the chip)

It is probably not the original one. Is there any link for the corect one? Or I can use some others? 

@rizacelik Thank you for your help!

BowenL1, 2025-02-27:
# status
INAV/STM32F411CE 6.1.1 Oct  7 2023 / 00:35:33 (GITDIR-N)
GCC-10.2.1 20201103 (release)
System Uptime: 54 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 8.02V (2S battery - OK)
CPU Clock=108MHz
STM32 system clocks:
  SYSCLK = 108 MHz
  HCLK   = 108 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=UNAVAILABLE, ACC=UNAVAILABLE, MAG=UNAVAILABLE, BARO=UNAVAILABLE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x20020000, Heap available: 1948
I2C Errors: 0, config size: 8997, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = ADC 4, used = none
System load: 12, cycle time: 1007, PID rate: 993, RX rate: 9, System rate: 9
Arming disabled flags: ANGLE CAL HWFAIL RX CLI
OSD: not enabled
VTX: not detected