rizacelik/STM32F411CEU6_INAV_Firmware #43: sbus support

octa54, 2024-02-24:
how can i connect sbus transmitter which you made to this transmitter what i have to select in ports pleas tell

rizacelik, 2024-02-24:
Follow the line with the blue arrow sign. Your connection should be like this.

[image]

octa54, 2024-02-28:
sir one more help in the receiver you used arduino nano in which TX pin is D0 but for my receiver I used arduino pro mini because its light weight and in it TX pin is D1 that's why I was not able to receive any data ,normal pwm channels are working properly so pleas help me with code to where I need to change pin.no from 0 to 1 in code it will help me a lot thanks

rizacelik, 2024-02-28:
Do not change this function **sbusPreparePacket**. This is the SBUS protocol. See other codes for changes.

[image]

The Arduino Pro Mini connected to (and powered by) an FTDI TTL will use Tx and RX pins, so terminate this connection after uploading the code. Give external power to the card and test it. Otherwise, it will not work because the Tx and RX pins will be in use.

octa54, 2024-02-28:
sir the main issue is that i have printed a PCB from this youtuber "howtomechatronics" design where he also made a transmitter and receiver 
he provided PCB gerber files and i printed those but he didn't provide SBUS protocal i tried to contact him he never replied i tried many ways to modify your sbus code and use for this specific circuit diagram as i said normal channels are working properly only problem is in sbus support pleas help me with this 
[image]
[image]

rizacelik, 2024-02-28:
You want me to adapt code for someone else's project. This is not possible. I really don't have time.
You should do this yourself by reviewing my codes. It is not hard. Adapt joystick data to Sbus protocol. My app has been tested and works without errors.
I try to help you as long as I have time, but I can't do more.
I hope you understand me.

octa54, 2024-03-02:
sir i tried my best but still not able to connect receiver to flight controller normal ppm signal is also not working