rizacelik/STM32F411CEU6_INAV_Firmware #44: One motor is getting hot

Karan1rao, 2024-03-04:
Recently I noticed that one motor (back left) is getting hot despite changing motor and esc, I wonder what could be the issue

rizacelik, 2024-03-04:
Which motor pin?

Karan1rao, 2024-03-15:
the one at the B6 pin

mursa1en, 2024-04-10:
I have also lost a motor and recently an esc using this pin (b6)

Karan1rao, 2024-04-10:
hoping rizacelik could solve this issue, as i have burnt two motors at B6 pins

rizacelik, 2024-04-10:
1. What is the ESC brand you use?
2. What protocols does it support?
3. And which protocol did you use in the installation?

Karan1rao, 2024-04-10:
1.  Hobbywing Skywalker-40A
2. My esc are standard esc.
3. I also use standard protocol

Karan1rao, 2024-04-10:
inav 7.0

rizacelik, 2024-04-10:
I will review the software again. I will try to fix this problem in INAV 7.1.

Karan1rao, 2024-04-10:
thanks brother!!!

mursa1en, 2024-04-10:
> 1. What is the ESC brand you use?
> 2. What protocols does it support?
> 3. And which protocol did you use in the installation?

I was using  Favourite Little bee 30A esc on oneshot 125 protocol and the motor was from dji phantom 3 (800 kv). 
I guess the esc lost was just an accident or co incident. But i lost the motor while setting up the FC. All of a sudden the motor started vibrating and  started to smoke

rizacelik, 2024-04-10:
If one or more of the engines is overheating, I recommend you take a look at the explanations in this video. Maybe your problem may be the explanation in this video.
https://www.youtube.com/watch?v=eo2sjTNS0pc