rizacelik/STM32F411CEU6_INAV_Firmware #45: W2812 led

Lalbabu2001, 2024-03-07:
Hi , when I was using version 6.1.0  the led strip used to work fine,
 In version 6.1.1 is not working.

suchigno, 2024-05-02:
Hi, in FW 7.1 I have the problem with the W2812 LED strip, if I enable the LED strip in the configuration, save and restart, the option remains disabled. I also tried with the CLI command. Software serials are disabled. I looked at the guide you posted but I can't find anything that explains why there could be a possible conflict. The pin for strip is A8.
Many thanks

rizacelik, 2024-05-03:
If the option in the image is active, it means there is support.

[image]

If the option in the image is not active, you need to turn on support with the CLI command. To do this, first run the CLI command "**feature**" and see the screen below. Is the **LED_STRIP** feature available on the screen? If so, it is active.

[image]

If the LED_STRIP feature is not on the screen, then use the command below.
CLI: Enable the LED_STRIP feature via the cli:

**feature LED_STRIP**

Run the CLI command "**feature**" again and see if the **LED_STRIP** feature is turned on on the screen. Check.

suchigno, 2024-05-03:
Hi, I did the check you asked for, you can see it in the image.
But when I save and reboot, the setting is disabled.
I don't know, it's very strange, everything works except the LED strip.
[image]

rizacelik, 2024-05-03:
There is a conflict somewhere. LED STRIP How did you connect it? If you have a video or photo, can you share it?

suchigno, 2024-05-03:
The LED strip is connected to the PA8 pin with a 300 ohm resistor.
The photo shows the breadboard with connections, the connected sensors are:
MPU6500
hmc5883
Buzzers
Led Strip
Receiver via SBUS and inverter (as per the connection shown by you). I also tried with PPM signal.
This all works.

I also post the port configuration, if there is a conflict, and the serial software are disabled I don't know what it could be, because I only use two sensors (SPI and i2c) and I don't use any serial modules such as GPS, telemetry etc.
However, even without the LED strip, it's still a nice project, I can even do without it and I use a LED with the buzzer pin to alert me to errors.
[image]
[image]

suchigno, 2024-05-03:
More than anything else I have an idea, but maybe I'll open another issue on this.
Would it be possible to connect a Bluetooth module, such as the HC-06, or jdy-08, to connect the board via Bluetooth, for example, to a phone and use the APP to calibrate it, change the parameters, without having to connect it to the laptop?

rizacelik, 2024-05-04:
There's a lot of wiring involved. To test LED STRIP, just connect the MPU9250 module to the board. If the LED STRIP does not activate again, there is another problem.

I don't know how to use a Bluetooth module like HC-06 or jdy-08. I've never seen such a connection option.

Please read source.

https://github.com/iNavFlight/inav/blob/master/docs/Telemetry.md
https://www.youtube.com/watch?v=ETtWE6A4Wo0
https://play.google.com/store/apps/details?id=com.eziosoft.ezgui.inav&hl=en_US

suchigno, 2024-05-04:
Thanks for your help, but the result doesn't change.
I removed all connections and connected only another IMU, (the MPU6050 i2C), and left the buzzer.
Unfortunately the strip configuration remains disabled after the reboot.
For bluetooth I decided not to implement it, I tried the OTG cable with my phone and with the speddybee APP and it works well. So for the first few flights I will use the phone to see if everything works well.

rizacelik, 2024-05-04:
If LEE STRIP is not working, there is nothing to do. Ignore this for now.

When you make the Bluetooth connection, write here how you did it. Let others benefit too. I can't keep up with everything.

suchigno, 2024-05-04:
Sure, my idea was to use a cheap bluetooth module (JDY-08) which should be compatible with the speedybee APP.
If everything works, post here how to do it.
Thanks

Harshhhh16, 2024-05-19:
i'm also facing same problem when i turned on led strip setup it automatically disabled and also cli is not working and i add 330ohm resistor still this problem facing