rizacelik/STM32F411CEU6_INAV_Firmware #46: Betaflight GPS

ArshaanTech, 2024-04-02:
Hey!
Nice work on all the firmware man! I really enjoy using them. I just can't get GPS to work with betaflight right now. I tried in the CLI and it just says unavailable. 
I think the potential solution can be that you build the firmware again with -DUSE GPS as Extra flags.  If it's possible please do it, i really need GPS working on betaflight! I tried on Inav and it does work flawlessly.
Thanks in advance!

ArshaanTech, 2024-04-13:
No possibilities? :(

rizacelik, 2024-04-13:
I recompiled it for you. Test this build.
[betaflight_4.5.0_STM32F411_STM32F411CE.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/14968221/betaflight_4.5.0_STM32F411_STM32F411CE.zip)

ArshaanTech, 2024-04-13:
Thank you so so much! Big help! will try it right away!

ArshaanTech, 2024-04-13:
It's working! GPS is now working on my done! Thank you so much, Haven't tested a flight but satellites are showing up. just a concern that the barometer is not showing up for some reason... is it supposed to be like that? can we not enable barometer and disable ppm and pwm instead for the space issue.
Thanks so much!

rizacelik, 2024-04-13:
Please provide some more detail when writing the problems. For example, the name of your barometer, is your barometer SPI connected? Information such as whether it is I2C connected or not is required.
Why did you need to disable ppm and pwm? explain this too.

ArshaanTech, 2024-04-13:
I am using a i2c connected barometer which is a BMP280. It is working with the older firmware but not the newer one (with GPS). I thought of disabling pwm and ppm to reduce the firmware size.

rizacelik, 2024-04-13:
Use CLI commands. It should be fixed.

set baro_bustype = I2C
set baro_i2c_device = 1

ArshaanTech, 2024-04-13:
Thank you! Got it working!