rizacelik/STM32F411CEU6_INAV_Firmware #47: Inav 7.0 GPS not working

athoxid, 2024-04-08:
hello Mr. Riza, i'm just upload firmware invav 7.0, but after setup GPS on port tab and configuration, GPS not working, the icon GPS on top of inav configurator is red color

rizacelik, 2024-04-09:
GPS works. Please check your cable connections. If you soldered it, repeat the soldering process.
Hundreds of people tested it and there were no problems.

athoxid, 2024-04-14:
GPS now working sir, previously i use inav 6.1.1 GPS working flawlessly but after updated to inav 7.0 GPS not working until just now i updated to inav 7.1 and change the wiring of RX1 TX1 GPS working flaswlessly again....Thank you for your great work!!! Happy Eid al-Fitr...Salam from Indonesia

closed 2024-04-14 by athoxid

pcgigabite65-cmd, 2026-09-06:
selamat @athoxid saya dari indonesia saya juga mengalami masalah di inav 6.1
setiap RTH terjun bebas, apakah pernah alami..apakah kita bisa komukasi lewat FB