rizacelik/STM32F411CEU6_INAV_Firmware #48: Yapılandırma Hakkında

SyberMafia, 2024-04-10:
Merhaba Rıza Bey,
inav_7.0.0_STM32F411CE_HEX dosyasını yüklediğimde mavi led 2 kez yanıp söner ve durur.
İnav tarafındanda tanınmıyor. Betaflight kısmında sorun yok.

GY-86 sensörüm mevcut. GY-87 ile aralarındaki tek fark barometrenin farklı olması.
İnav da aynı konfigurasyonlar ile hex dosyasını oluşturamıyorum. mpu6050 inav5.1.1 den sonraki sürümler de kaldırılmış sadece spi olarak kullanılabiliyor.
Betaflight içinde durum aynı. yapmış olduğun veya sonradan eklediğin dosyalarıda paylaşbilir misin?
betaflightta GY-87 sensörünü I2C üzerinden nasıl yapılandırdın. config.h dosyasını paylaşabilir misin?

BETAFLIGHT 4.5.0 için aşağıdaki gibi yapılandırdığımda sadece MS5611 Barometre sensörü gözüküyor
İvme gyro ve pusula sensörlerini bulamıyor.

#define USE_GYRO
#define USE_GYRO_MPU6050

#define USE_ACC
#define USE_ACC_MPU6050

#define USE_MAG
#define USE_MAG_HMC5883

#define USE_BARO
#define USE_BARO_MS5611

// UART & I2C
#define USE_I2C
#define I2C1_SCL_PIN         PB8
#define I2C1_SDA_PIN         PB9
#define UART1_TX_PIN         PA9
#define UART1_RX_PIN         PA10
#define UART2_TX_PIN         PA2
#define UART2_RX_PIN         PA3
#define SOFTSERIAL1_TX_PIN   PB3
#define SOFTSERIAL1_RX_PIN   PB10

#define BARO_I2C_INSTANCE    (I2CDEV_1)
#define MAG_I2C_INSTANCE     (I2CDEV_1)

rizacelik, 2024-04-10:
Merhaba ilker bey;
inav_7.0.0_STM32F411CE_HEX dosyasını yüklemeden önce STM32F411CE kartınızın üzerinde mutlaka gy87 modülünün bağlantılarını yapmış olmanız gerekir. Modül olmadan bağlantı sağlanmıyor.  MPU9250 kullanmanızı tavsiye ederim. Sorunsuz çalışacaktır.

NOT: MPU6050 modülü CPU yükünü %90 çıkardığından INAV bunu kaldırdı. Betaflight içinde aynı durum var. CPU yükünü aşırı yükseltmekte. Kullanmamanızı tavsiye ederim. CPU yükü fazla olduğunda ARM işlemi gerçekleşmez. En iyisi SPI ara yüzü ile kullanılan MPU9250 modüldür. Bir tane de ek barometre kullanırsınız. 

INAV 7.1 versiyonu STM32F411CE kartı için son destekti. Ardık bu kart INAV için kullanılmayacak. Betaflight için destek devam edecek.

Betafilght için aşağıdaki config.h dosyasını aşağıdaki gibi yapılandırdım.

#pragma once

#define FC_TARGET_MCU     STM32F411

#define BOARD_NAME        STM32F411CE
#define MANUFACTURER_ID   ST41

#define USE_UNIFIED_TARGET
#define USE_GYRO
#define USE_GYRO_SPI_MPU6000
#define USE_GYRO_SPI_MPU6500
#define USE_GYRO_SPI_MPU9250
#define USE_GYRO_MPU6050
#define USE_ACC
#define USE_ACC_SPI_MPU6000
#define USE_ACC_SPI_MPU6500
#define USE_ACC_SPI_MPU9250
#define USE_ACC_MPU6050
#define USE_BARO_BMP280
#define USE_BARO_MS5611
#define USE_BARO_BMP085
#define USE_MAG_HMC5883
#define USE_MAG_QMC5883
#define USE_MAX7456

#define MOTOR1_PIN           PB4
#define MOTOR2_PIN           PB5
#define MOTOR3_PIN           PB6
#define MOTOR4_PIN           PB7
#define MOTOR5_PIN           PA5
#define MOTOR6_PIN           PB10
#define SERVO1_PIN           PA0
#define SERVO2_PIN           PA1

#define RX_PPM_PIN           PA3
#define LED_STRIP_PIN        PA8
#define UART1_TX_PIN         PA15
#define UART1_RX_PIN         PB3
#define UART2_TX_PIN         PA2
#define UART2_RX_PIN         PA3
#define I2C1_SCL_PIN         PB8
#define I2C1_SDA_PIN         PB9
#define LED0_PIN             PC13
#define SPI2_SCK_PIN         PB13
#define SPI2_SDI_PIN         PB14
#define SPI2_SDO_PIN         PB15
#define ADC_VBAT_PIN         PB0
#define ADC_CURR_PIN         PB1
#define MAX7456_SPI_CS_PIN   PC14
#define GYRO_1_EXTI_PIN      PA7
#define GYRO_2_EXTI_PIN      NONE
#define GYRO_1_CS_PIN        PB12
#define USB_DETECT_PIN       PC15

#define TIMER_PIN_MAPPING \
    TIMER_PIN_MAP( 0, PA3 , 3, -1) \
    TIMER_PIN_MAP( 1, PB4 , 1,  0) \
    TIMER_PIN_MAP( 2, PB5 , 1,  0) \
    TIMER_PIN_MAP( 3, PB6 , 1,  0) \
    TIMER_PIN_MAP( 4, PB7 , 1,  0) \
    TIMER_PIN_MAP( 5, PA5 , 1,  0) \
    TIMER_PIN_MAP( 6, PB10, 1,  0) \
    TIMER_PIN_MAP( 7, PA0 , 2,  0) \
    TIMER_PIN_MAP( 8, PA1 , 2,  0) \
    TIMER_PIN_MAP( 9, PA2 , 2,  0) \
    TIMER_PIN_MAP( 10, PA8 , 1,  0)

#define ADC1_DMA_OPT        1

#define DEFAULT_DSHOT_BURST DSHOT_DMAR_AUTO
#define DEFAULT_DSHOT_BITBANG DSHOT_BITBANG_OFF
//#define USE_I2C
//#define USE_I2C_DEVICE_1
//#define I2C_DEVICE I2CDEV_1
#define USE_BARO
#define BARO_I2C_INSTANCE I2CDEV_1
//#define DEFAULT_BARO_DEVICE BARO_NONE
#define MPU_I2C_INSTANCE I2CDEV_1
#define USE_MAG
#define USE_MAG_DATA_READY_SIGNAL
#define MAG_I2C_INSTANCE I2CDEV_1
#define DEFAULT_CURRENT_METER_SOURCE CURRENT_METER_ADC
#define DEFAULT_VOLTAGE_METER_SOURCE VOLTAGE_METER_ADC
//TODO #define VBAT_DETECT_CELL_VOLTAGE 300
//#define BEEPER_INVERTED
#define SYSTEM_HSE_MHZ 25
#define MAX7456_SPI_INSTANCE SPI2
#define USE_SPI_GYRO
#define GYRO_1_SPI_INSTANCE SPI2
#define GYRO_1_ALIGN CW0_DEG

rizacelik, 2024-04-10:
Betaflight da MPU6050 yine de görmez ise o zaman aşağıdaki CLI komutlarını kullanın.

set mag_bustype = I2C
set mag_i2c_device = 1
set baro_bustype = I2C
set baro_i2c_device = 1
set gyro_1_bustype = I2C
set gyro_1_i2cBus = 1

rizacelik, 2024-04-11:
Bir kaç tane GY-91 aliexpress'ten aldım ama sahte çıktı. Üzerinde MPU6500 ve BMP280 var. Pusula eksik. Sizde MPU9250 veriyor mu? yani INAV bu sensörü MPU9250 olarak görüyor mu?

SyberMafia, 2024-04-11:
inav7 de tekrar denemedim ama betaflightta çalışıyor
inavda ilk mesajda belirttiğiniz gibi deneyeceğim tekrar.
Betaflightta: Gyro MPU9250 Acc MPU9250 MAG AK8963  olarak görünüyor. Bende aliexpresten almıştım.

Aslında sensör MPU9255, pyton ile WHO_AM_I 0x75 adresini hex olarak okuduğumda 0x73 değerini veriyor
[image]

#define MPUx0x0_WHO_AM_I_CONST              (0x68) // MPU3050, 6000, 6500 and 6050
#define MPU6500_WHO_AM_I_CONST              (0x70)
#define MPU9250_WHO_AM_I_CONST              (0x71)
#define MPU9255_WHO_AM_I_CONST              (0x73)

[https://github.com/iNavFlight/inav/blob/master/docs/assets/images/Blackpill_F411.png](url)
INAV da spi1 MPU için tanımlanmış Benimde spi1 de 8MB spi flash var. Onun için özel derleme yapmak gerekiyor.
CLI kısmı açılsa oradan değiştirebilirm belki ama flashladığımda pc bile görmüyor.

Tekrardan derleyip durumu burada paylaşırım

SyberMafia, 2024-04-11:
Malesef Inav için yaptığım veya hazır olan hex dosyalarını karta yüklediğimde usb tanınmıyor.
Betaflight 4.5.0 için Barometreyi ayarlayamamıştım. Şimdi GY-91 tam bir şekilde çalışıyor.
Ekran görüntüsünden sensör bilgilerine bakabilirsiniz.
[image]

Betaflight 4.5.0 için CONFIG.H dosyası bu şekilde Belki INAV için işinize yarayabilir.

#pragma once

#define FC_TARGET_MCU     STM32F411

#define BOARD_NAME        MYDIYDRONE2
#define MANUFACTURER_ID   MYT411

#define USE_GYRO
#define USE_GYRO_SPI_MPU9250
#define GYRO_1_CS_PIN        PB12
#define GYRO_1_SPI_INSTANCE  SPI2

#define USE_ACC
#define USE_ACC_SPI_MPU9250

#define USE_MAG
#define USE_MAG_MPU925X_AK8963

#define USE_BARO
#define USE_BARO_SPI_BMP280
#define BARO_CS_PIN          PA13
#define BARO_SPI_INSTANCE    SPI2

// UART & I2C
#define I2C_DEVICE           I2CDEV_1
#define I2C1_SCL_PIN         PB8
#define I2C1_SDA_PIN         PB9
#define UART1_TX_PIN         PA9
#define UART1_RX_PIN         PA10
#define UART2_TX_PIN         PA2
#define UART2_RX_PIN         PA3
#define SOFTSERIAL1_TX_PIN   PB3
#define SOFTSERIAL1_RX_PIN   PB10

// MOTORLAR
#define MOTOR1_PIN           PB4
#define MOTOR2_PIN           PB5
#define MOTOR3_PIN           PB6
#define MOTOR4_PIN           PB7

// KUMANDA
#define USE_RX_PPM
#define RX_PPM_PIN           PA3
#define DEFAULT_RX_TYPE      RX_TYPE_PPM
#define SERIALRX_PROVIDER    SERIALRX_SBUS
#define SERIALRX_UART        SERIAL_PORT_USART2

// LED ve SERIT LED
#define LED0_PIN             PC13
#define LED_STRIP_PIN        PA8

// SPI BAGLANTI
#define SPI1_SCK_PIN         PA5
#define SPI1_SDI_PIN         PA6
#define SPI1_SDO_PIN         PA7

#define SPI2_SCK_PIN         PB13
#define SPI2_SDI_PIN         PB14
#define SPI2_SDO_PIN         PB15

// BLACKBOX
#define USE_FLASH
#define FLASH_CS_PIN         PA4
#define FLASH_SPI_INSTANCE   SPI1
#define USE_FLASH_M25P16
#define USE_FLASH_W25M
#define DEFAULT_BLACKBOX_DEVICE     BLACKBOX_DEVICE_FLASH

// ADC
#define ADC_VBAT_PIN         PB0
#define ADC_CURR_PIN         PB1
#define ADC_INSTANCE         ADC1
#define ADC1_DMA_OPT         0
#define DEFAULT_CURRENT_METER_SOURCE CURRENT_METER_ADC
#define DEFAULT_VOLTAGE_METER_SOURCE VOLTAGE_METER_ADC

// DSHOT
//#define DEFAULT_DSHOT_BURST DSHOT_DMAR_ON

// PINIO PIN
#define PINIO1_PIN           PA14
#define PINIO1_CONFIG        129
#define PINIO1_BOX           40

#define SYSTEM_HSE_MHZ       25

// TIMERS & DMA
#define TIMER_PIN_MAPPING \
    TIMER_PIN_MAP( 0, PA3 , 3, -1) \
    TIMER_PIN_MAP( 1, PB4 , 1,  0) \
    TIMER_PIN_MAP( 2, PB5 , 1,  0) \
    TIMER_PIN_MAP( 3, PB6 , 1,  0) \
    TIMER_PIN_MAP( 4, PB7 , 1,  0) \
    TIMER_PIN_MAP( 5, PB3 , 1,  0) \
    TIMER_PIN_MAP( 6, PB10, 1,  0) \
    TIMER_PIN_MAP( 7, PA2 , 2,  0) \
    TIMER_PIN_MAP( 8, PA8 , 1,  0)

Betaflight da yeni model oluşturmak kolay. Inav için bunu belki doğru yapamadım bilmiyorum.
INAV da yeni model nasıl tanımlıyorsunuz ?
Betaflight da configs klasörünün içine yeni bir klasör oluşturarak içinede config.h dosyası oluşturuyorum
make KARTADI deyince bir sorun yoksa hızlı bir şekilde hex dosyasını oluşturuyor.
Inav biraz karışık geldi bana

rizacelik, 2024-04-12:
CMakeLists.txt Dosyasında 25MHZ kristal tanımlaması var mı? eğer yoksa saat hızı doğru ayarlanamadığı için USB için gerekli 48MHZ ayarlanamadığı için USB desteği olmayacaktır. Onun dışında başka bir değişiklik yok.

target_stm32f411xe(STM32F411CE HSE_MHZ 25 SKIP_RELEASES)

Bendeki GY91 modüllerini test ettim, hepsi maalesef sahte. Aliexpress te Aldığınız yerin linki varsa paylaşabilir misiniz. Almak isterim.

SyberMafia, 2024-04-12:
İnav yeni hedef nasıl oluşturuluyor bulamadığım için BLACKPILL_411 hazır bir kart üzerinden değişiklik yapmıştım belki ordan bir sıkıntı çıkıyor. Betaflight gibi dökümantasyonları yok.
target_stm32f411xe(BLACKPILL_411_BLACKBOX HSE_MHZ 25 SKIP_RELEASES) tanımlaması mevcut

Aliexpress mağazası boş görünüyor ama satıcı aynı
https://www.tztstore.com/goods/show-5620.html

SyberMafia, 2024-04-12:
Bu hex dosyasında sadece jiroskobu gördü. ayarlardan manuel olarak hepsini otomatik yaptım. sensörü tam algıladı.
Gönderdiğiniz hex dosyasını flashladıktan sonra usb de otomatik olarak algılandı. Sorunsuz.
Sensör bağlanmadan inav programı karta bağlanmıyor :) Dediğiniz gibi sensörü takınca connect oluyor.
SPI FLASI algılamadı. Benim kullandığım ayarları ekliyorum. rica etsem bu son çalışan ayarlarınız paylaşabilir misiniz?
[BPMYDRONEV1.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/14961385/BPMYDRONEV1.zip)
[image]
[image]

rizacelik, 2024-04-12:
Dosyalarınızda değişiklik yaptım. Derleyip test edebilirsiniz.
[BPMYDRONEV1.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/14963461/BPMYDRONEV1.zip)

SyberMafia, 2024-04-12:
INAV dan resmen nefret ettim derlemesi Betaflighta göre çok yavaş ve uzun sürüyor.
Ama daha stabil ve geliştirilmiş. Son versiyonda tanımlama özelliklerinde baya değişiklik olmuş.
İkisinde de nasıl sıfırdan anakart oluşturacağımı uzun uğraşlar sonucunda sizinde yardımınızla çözmüş bulunmaktayım.
Bağlantılarda veya USB kablosunda bir temassızlık olduğunda otomatik olarak DFU moduna giriyor. (INAV için)
Betafligth şuan daha sorunsuz gibi görünüyor, INAV için biraz daha beklenebilir.
INAV derlerken beni deli ediyor aşırı yavaş kalıyor. Betaflight 1dk da derleniyor ve ayarlar sorunsuz geliyor.
Inavda birçok eski özellikler desteğini yitirmiş durumda şuan 7.1.0 da ppm bile yok. Betaflight 4.5.0 da mevcut.
[DIYDRONE_V1.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/14963920/DIYDRONE_V1.zip)
sizden önce bu dosyayı derlemiştim ve şuan GY-91 sensörü için çalışıyor INAVda :)
WeAct STM32H743 bu kartın siparişi verdim Linki Burada [https://tr.aliexpress.com/item/1005006632336183.html](url)
480 MHZ çalışma hızı, 2MB Programlanabilir Hafızası 8MB extra SPI flashı ve hafıza kartı için dahili yeri mevcut.
bu kart içinde uygun konfigürasyonlar ayarlanabiliyor, uçuş kontrol kartı almaktan daha uygun fiyata geliyor hem.
Henüz uçuş denemesi yapmadım. Büyük bir ihtimal yeni kart geldiğinde deneme yapacağım.
Yardımlarınız için teşekkür ederim.

closed 2024-04-14 by SyberMafia