rizacelik/STM32F411CEU6_INAV_Firmware #49: Help to build Weact H743 firmware.

Lynklee321, 2024-04-13:
I am getting acquainted with inav on the H743 circuit board from weact, and am having problems receiving gyro (BMI160). I tried with the servo on motor 1 with the default settings and the servo vibrates and gets hot but can still rotate. So I think I have a problem with the HSE. I researched and corrected this section in the Cmakelist.txt file but it did not fix the problem (target_stm32h743xi(H743 HSE_MHZ 25).

rizacelik, 2024-04-13:
Let's compress the configuration files and put them here. You need to look at the file.

Lynklee321, 2024-04-13:
diff all

# version
# INAV/FOXEERH743 7.0.0 Apr 13 2024 / 13:21:03 (GITDIR-N)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# Timer overrides
timer_output_mode 2 SERVOS
timer_output_mode 3 SERVOS
timer_output_mode 7 SERVOS
timer_output_mode 11 SERVOS

# Outputs [servo]

# safehome

# features

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# Receiver: Channel map

# Ports

# LEDs

# LED color

# LED mode_color

# Modes [aux]

# Adjustments [adjrange]

# Receiver rxrange

# temp_sensor

# Mission Control Waypoints [wp]
#wp 0 invalid

# OSD [osd_layout]

# Programming: logic

# Programming: global variables

# Programming: PID controllers

# master
set looptime = 500
set gyro_main_lpf_hz = 110
set gyro_main_lpf_type = PT1
set gyro_use_dyn_lpf = ON
set gyro_dyn_lpf_min_hz = 85
set gyro_dyn_lpf_max_hz = 300
set gyro_dyn_lpf_curve_expo = 3
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 100
set setpoint_kalman_q = 200
set acc_hardware = BMI160
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = NONE
set blackbox_rate_denom = 2
set motor_pwm_protocol = STANDARD
set failsafe_procedure = DROP
set applied_defaults = 2
set airmode_type = THROTTLE_THRESHOLD

# mixer_profile
mixer_profile 1

set platform_type = AIRPLANE
set has_flaps = ON
set model_preview_type = 28

# Mixer: motor mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# Mixer: servo mixer
smix reset

smix 0 1 0 100 0 -1
smix 1 2 0 100 0 -1
smix 2 3 1 50 0 -1
smix 3 3 2 -50 0 -1
smix 4 4 1 -50 0 -1
smix 5 4 2 -50 0 -1

# mixer_profile
mixer_profile 2

# Mixer: motor mixer

# Mixer: servo mixer

# profile
profile 1

set mc_i_pitch = 90
set mc_d_pitch = 27
set mc_p_roll = 36
set mc_i_roll = 80
set mc_d_roll = 25
set mc_p_yaw = 35
set mc_i_yaw = 80
set dterm_lpf_type = PT3
set d_boost_min =  1.000
set d_boost_max =  1.000
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set smith_predictor_delay =  1.500
set tpa_rate = 20
set tpa_breakpoint = 1200
set rc_expo = 75
set rc_yaw_expo = 75
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# profile
profile 2

# profile
profile 3

# battery_profile
battery_profile 1

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
mixer_profile 1
profile 1
battery_profile 1

# save configuration
save

#

Lynklee321, 2024-04-13:
For the configuration, I used FOXEERH743 as a reference.

Lynklee321, 2024-04-13:
[target.txt](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/14965859/target.txt)

rizacelik, 2024-04-13:
https://github.com/WeActStudio/MiniSTM32H7xx/blob/master/HDK/STM32H7xx%20SchDoc%20V11.pdf
There is a board diagram at this address. Check if the pins comply with the diagram in the configuration file you are using by looking at this diagram. If there are pins used, remove them from the configuration file. If there is a pin used in the schematic and you used it for another function in the configuration file, your compilation will be problematic.
Since I don't have much time, I can't review them one by one, but you can do it.

rizacelik, 2024-04-13:
You can use LCD and Camera pins in this diagram. Problems will occur with other pins used for other purposes in the schematic. Examine and configure the pins in your file accordingly.
Too many pins on this board are used for other purposes. Frankly, there are very few pins left.

Lynklee321, 2024-04-13:
I checked the pins for duplicates, basically I just used spi1 or 4 to check the gyro to see if they work or not.

rizacelik, 2024-04-13:
Use these pins for SPI1
[image]

Lynklee321, 2024-04-13:
Does it have a problem with the flash on the circuit?

Lynklee321, 2024-04-13:
[image]

rizacelik, 2024-04-13:
The top part of the target.h file should look like this.

```
#pragma once

#define TARGET_BOARD_IDENTIFIER "F743"
#define USBD_PRODUCT_STRING "FOXEERH743"

#define USE_TARGET_CONFIG

#define LED0 PE3

// #define BEEPER PD2
// #define BEEPER_INVERTED

// *************** SPI1 IMU0 MPU6000 ****************
#define USE_SPI
#define USE_SPI_DEVICE_1
#define SPI1_SCK_PIN            PB3
#define SPI1_MISO_PIN           PB4
#define SPI1_MOSI_PIN           PD7

#define USE_IMU_MPU6000
#define IMU_MPU6000_ALIGN CW180_DEG
#define MPU6000_SPI_BUS BUS_SPI1
#define MPU6000_CS_PIN PD6

#define USE_IMU_BMI270
#define IMU_BMI270_ALIGN CW180_DEG
#define BMI270_SPI_BUS BUS_SPI1
#define BMI270_CS_PIN PD6
```

Your target.c file should be like this

```
#include <stdint.h>

#include "platform.h"

#include "drivers/bus.h"
#include "drivers/io.h"
#include "drivers/pwm_mapping.h"
#include "drivers/timer.h"
#include "drivers/pinio.h"
#include "drivers/sensor.h"

timerHardware_t timerHardware[] = {

    DEF_TIM(TIM3, CH2, PB5,     TIM_USE_OUTPUT_AUTO, 0, 1),   // S1
    DEF_TIM(TIM3, CH3, PB0,     TIM_USE_OUTPUT_AUTO, 0, 2),   // S2
    DEF_TIM(TIM3, CH4, PB1,     TIM_USE_OUTPUT_AUTO, 0, 3),   // S3

    DEF_TIM(TIM15, CH1, PE5,    TIM_USE_OUTPUT_AUTO, 0, 0),   // S4

    DEF_TIM(TIM1,  CH1, PA8,  TIM_USE_LED, 0, 9),    // LED_2812
};

const int timerHardwareCount = sizeof(timerHardware) / sizeof(timerHardware[0]);
```

The pins of the IMU sensor have changed. Some of the motor pins were removed. Test it by compiling it and see the result.

rizacelik, 2024-04-13:
Delete the config.c file, do not use it.

Lynklee321, 2024-04-13:
It still doesn't solve the problem, I think it is necessary to find out the reason why the servo vibrates when not controlled, perhaps the signal is disturbed causing the mcu to not be able to detect the gyro.

rizacelik, 2024-04-13:
My advice to you is that if you have modules such as MPU9250 or MPU6000, MPU6500, you should test it with one of them. If there is a problem with these modules, then we can say that there is a hardware problem with the MCU.
That's the only way to find out.

Also change your servo. It may also be caused by the servo. If you are using ESC, it may also be related to ESC.
These are my assumptions.

SyberMafia, 2024-04-13:
Hi Lynklee321,
The internal crystal of your card is 48MHZ.
In this way, the signal is distorted.
Remove the target_stm32h743xi(H743 HSE_MHZ 25) "HSE_MHZ 25" section from here.
correct -> **target_stm32h743xi(H743)**
The STM32F411 cards we use are available in 2 types. 8Mhz and 25MHZ crystal. Since our current cards are 25MHZ, we set it this way. There is no need for such a setting for the card you are using.
If your problem persists, you need to test your servo with another card or try another servo.

Lynklee321, 2024-04-13:
> My advice to you is that if you have modules such as MPU9250 or MPU6000, MPU6500, you should test it with one of them. If there is a problem with these modules, then we can say that there is a [...]

i also tried with mpu 6500

Lynklee321, 2024-04-13:
> Hi Lynklee321, The internal crystal of your card is 48MHZ. In this way, the signal is distorted. Remove the target_stm32h743xi(H743 HSE_MHZ 25) "HSE_MHZ 25" section from here. correct -> **target_stm32h743xi(H743)** - replace and recompile. The STM32F411 cards we use are available in 2 types. 8Mhz and 25MHZ crystal. Since our current cards are 25MHZ, we set it this way. There is no need for such a setting for the card you are using. If your problem persists, you need to test your servo with another card or try another servo.

When I install with 25mhz, in the status tab it shows that the mcu is running at 500Mhz, so is there an error or not? in the circuit diagram and when looking at the actual circuit I see that there is an external 25mhz crystal.

SyberMafia, 2024-04-14:
> When I install with 25mhz, in the status tab it shows that the mcu is running at 500Mhz, so is there an error or not? in the circuit diagram and when looking at the actual circuit I see that there is an external 25mhz crystal.

You are right, there is a 25 MHZ crystal available externally.
Operating Frequency appears to be Max 480 MHZ.
I looked at all the settings files created with the STM32H743.
I have not come across a configuration file that uses 25mhz other than F4 serial cards.
You cannot plug a servo into a channel configured for ESC. Since it is configured according to Dshot, its operating frequencies are different. While the servo is driven with 50hz PWM, Dshot gives an output between 300 - 600 Hz for ESC.
[image]
Could you try adjusting the esc and servo outputs like this to test them? I can't think of anything else right now.

Lynklee321, 2024-04-14:
> > When I install with 25mhz, in the status tab it shows that the mcu is running at 500Mhz, so is there an error or not? in the circuit diagram and when looking at the actual circuit I see that there is an external 25mhz crystal.
> 
> You are right, there is a 25 MHZ crystal available externally. Operating Frequency appears to be Max 480 MHZ. I looked at all the settings files created with the STM32H743. I have not come across a configuration file that uses 25mhz other than F4 serial cards. You cannot plug a servo into a channel configured for ESC. Since it is configured according to Dshot, its operating frequencies are different. While the servo is driven with 50hz PWM, Dshot gives an output between 300 - 600 Hz for ESC. ![Ekran görüntüsü 2024-04-14 090056](https://private-user-images.githubusercontent.com/69259313/322250808-d681e71f-ee4d-403f-a56c-c65b8ab32aa3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMwNzU3MzEsIm5iZiI6MTcxMzA3NTQzMSwicGF0aCI6Ii82OTI1OTMxMy8zMjIyNTA4MDgtZDY4MWU3MWYtZWU0ZC00MDNmLWE1NmMtYzY1YjhhYjMyYWEzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE0VDA2MTcxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA2MGY3MmFkNmE4ODJjZjdkNTk4NzhlN2JiZGIxMTU0YTNjNDVhOWRjOTExZjBjMGQ1NjM3OTM5ZDdlYmJlZTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.OrqYjG6YIEB1UcGGdzIpVtGNrADle1geLzMy_8hKMn0) Could you try adjusting the esc and servo outputs like this to test them? I can't think of anything else right now.

I know that, I also tried switching to plane mode to test sevor but there was no difference

rizacelik, 2024-04-14:
What's the screenshot of your Plane configuration's Mixer settings?
There is more than one servos in the plane. Is this one of the servos? Or in all of them? have the same problem.

Lynklee321, 2024-04-14:
Haahah  my bad, I tried everything from changing the 8mhz quartz oscillator, removing the qspi flash. Ultimately my servo failed. As for the motor, it has been resolved. As for the gyro, nothing has changed yet. If you don't mind, can you build the fw for me with a frequency of 8mhz (defaut) with the taget file as above. Thanks for your help.

rizacelik, 2024-04-14:
There is a 25mhz crystal on the board. Are you saying you changed it to 8MHZ?
[image]

Lynklee321, 2024-04-14:
> There is a 25mhz crystal on the board. Are you saying you changed it to 8MHZ? [...]

yes, blow it out and replace it with 8mhz

rizacelik, 2024-04-14:
Didn't you make the previous compilations? You can compile this yourself.