rizacelik/STM32F411CEU6_INAV_Firmware #50: cannoct connect to inav configurator with inav 7.0+

Dex94, 2024-04-14:
I tried to connect the board with inav configurator but I can connect it if I flash maximum 6.1.1 version, with 7 and 7.1 one I can't

rizacelik, 2024-04-14:
As can be seen in the diagram, you need to connect one MPU9250 or MPU6050. Otherwise, the connection will not occur.

Dex94, 2024-04-14:
> As can be seen in the diagram, you need to connect one MPU9250 or MPU6050. Otherwise, the connection will not occur.

Is there a way to edit the target to test inav without connecting gyro? I also tried to compile blackpill_f411 but the result is the same.

Dex94, 2024-04-14:
> This is the flight control card. There is no other way.

in inav 6.1 I can connect the board without any sensor. Has anything changed in the inav repo? It seems to be a security layer that it doesn't start without the gyro

rizacelik, 2024-04-14:
You can do whatever you want in inav 6.1.
Things may change in versions. In these latest versions you need to connect one sensor.

closed 2024-04-14 by Dex94

hendrachastinaputra, 2024-04-20:
> As can be seen in the diagram, you need to connect one MPU9250 or MPU6050. Otherwise, the connection will not occur.

Sorry, but I've connected MPU6050 and i can connect it with INAV 7.0, but I can't connect it with INAV 7.1. Should I use MPU9250 for INAV7.1?
Thanks

rizacelik, 2024-04-20:
yes

hendrachastinaputra, 2024-04-20:
> yes

oke, thanks sir

suchigno, 2024-05-01:
HI, very very great work!
But I have a problem with fw 7.1.0.
If I install firmware 6.1.0 (with INAV 6.1), the F411 sees all the sensors correctly (MPU6050, BMP280, HMC5883, buzzer etc.)
I made a PCB for the wire of version 7.1.0, and after having correctly flashed the F411 with 7.1.0 , INAV 7.1  does not connect to the board via COM. However, it doesn't tell me the error why it doesn't connect. Can you tell me what the reason could be? (I also used full chip erase).
Many thanks

rizacelik, 2024-05-01:
There may be a fault in the PCB. Try the board without PCB. It must have an IVE sensor on it. Otherwise it will not connect.
Here we have it running INAV7.1. https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/issues/51#issuecomment-2088421103

suchigno, 2024-05-01:
Hi, sorry, I was wrong...
I haven't soldered anything on the PCB yet, I was testing first on a breadboard. To flash the micro I removed it from the breadboard, and tried to connect it without reinserting it into it.
I read in one of your posts that a gyroscope/accelerometer had to be connected to it.
Now works :-)
Thank you

jpfree, 2025-03-03:
i flash the 7.00 ,and wire a mpu9250/6500,and i check the wire is ok, but still can't connected .

[image]

suchigno, 2025-03-03:
If you have tested the connections on the pcb and they work, maybe the pcb is not the problem. Install FW 7.1 not 7.0, and use the 7.1 configurator