rizacelik/STM32F411CEU6_INAV_Firmware #51: sonar

Harshhhh16, 2024-04-15:
hello brother i want to add 
1.  VL53L0X TOF Based LIDAR Laser Distance Sensor in this microcontroller
2.  5v Buzzer
3.  servo gimbal
4.  when i connect ublox 6m gps to 5v its led not blinking and heat up what is problem there ? is it 3.3v ??

rizacelik, 2024-04-15:
1. VL53L0X ToF Distance Ranging Sensor is not supported. All pins are in use. There is no extra pin.
2. For 5v Buzzer, you can connect it instead of CURRENT_METER_ADC in INAV 7.1.0 version. So PB1 pin.
3. You can configure servo gimbal support from Mixer settings.
4. You may have connected the + to - polarities incorrectly on Ublox GPS. Normally it is not possible to warm up. Double check and be careful. Do not burn GPS.

Harshhhh16, 2024-04-15:
Can i directly connect to Port B1 or any resistor needed between them?
And my gps module is dead or it csn be repaired?

rizacelik, 2024-04-16:
You can connect a 100ohm resistor to Port B1.
I can't say anything about GPS. You will check it yourself.

Harshhhh16, 2024-04-16:
Hello
Can i use this type of GPS is there is any problem ?? or what is difference between this module and  without On-board ceramic antenna module ??
GPS NEO-7M Satellite Positioning Module Development Board for Arduino STM32 51
[image]

rizacelik, 2024-04-16:
You can use any GPS module you want. 
There is no difference between them. Only some GPS modules become ready more quickly. So it gets the data sooner. GPS NEO modules initially receive satellite information in a longer time, but there is a slight difference with other modules afterwards.
You can learn the difference between these by examining the datasheet.

Harshhhh16, 2024-04-17:
What is difference between inav 7.1 vs 6.1 is their any major changes?

rizacelik, 2024-04-17:
Of course, there are differences between INAV versions. New features are added or removed.
https://github.com/iNavFlight/inav/releases/tag/7.1.0
For a 68K resistor, you can connect more than one resistor in series. Make a calculation to complete 68K by looking at the resistor catalogs in your country.

For any other questions, please refer to the diagram and descriptions. You can find all the information there.

Harshhhh16, 2024-04-17:
hello can you send quadcopter detail diagram with each and every component on board on latest version of inav ??
because i want to know where to connect black box and some other sensors you provide diagram is okay but i want all component on board embedded diagram....

rizacelik, 2024-04-17:
I don't have enough time and I try to help you when I have time.
All components you can use with this board are given in the description section. There is no additional information. Please review it thoroughly. How to connect all the components and which pins to use are given. I don't know how else I can help.

Harshhhh16, 2024-04-22:
Hello
i connect my 7N-0-002 Gps and also all connection are correct and i wait almost 1+hrs for gps signal but still gps not connected with satellite what
[image]
's the problem ?? and i'm also in open space but still not connected :(

Harshhhh16, 2024-04-22:
Hello
i connect my 7N-0-002 Gps and also all connection are correct and i wait almost 1+hrs for gps signal but still gps not connected with satellite what's the problem ?? and i'm also in open space but still not connected :(

[image]
[image]

rizacelik, 2024-04-22:
How did you connect the pins?
GPS pin -> Board pin
Why use INAV 6.0? Have you tried the current version?

rizacelik, 2024-04-22:
"NEO GPS" modules acquire data after 15 to 25 minutes on first use. Wait for this period. It will get your location information.

mursa1en, 2024-04-26:
> Hello
> i connect my 7N-0-002 Gps and also all connection are correct and i wait almost 1+hrs for gps signal but still gps not connected with satellite what's the problem ?? and i'm also in open space but still not connected :(
> 
> 
> 
> 
> 
> [image]
> [image]
> 

This GPS module doesn't connect to satellites in indoor conditions. If you're inside your home try to keep the gps near the window if it doesn't solve the issue then try to keep the gps under the open sky.

You can access the inav configurator on the "Speedybee" android app.

mursa1en, 2024-04-27:
> 1. VL53L0X ToF Distance Ranging Sensor is not supported. All pins are in use. There is no extra pin.
> 2. For 5v Buzzer, you can connect it instead of CURRENT_METER_ADC in INAV 7.1.0 version. So PB1 [...]

I guess range finders like the vl53l1x/vl53l0x uses i2c to communicate with the flight controller. 

Isn't it possible to utilize the range finder over the existing i2c pins? 

I tried to connect a vl53l1x to the board but it didn't work. Inav (7.0)  didn't detect the module. 
Is the stm32f411ceu6 not capable enough or the firmware doesn't have support for this module?

rizacelik, 2024-04-27:
Added vl53l0x support. You will use I2C1 pins. You can test it. Write your result here.
[inav_7.1.0_STM32F411CE.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/15138123/inav_7.1.0_STM32F411CE.zip)

mursa1en, 2024-04-27:
Sorry, I accidentally fried my only F411 board with a wrong polarity hookup. So now I can’t test it out. I’ve ordered some more, but it’ll be like two weeks before they show up where I live. and I’ve got the VL53L1X sensor module. I don't have the VL53L0X module.

rizacelik, 2024-04-27:
vl53l1x/vl53l0x support both. No problem. Support added. You test at the appropriate time.

Harshhhh16, 2024-04-29:
I want mission planner software for talemetry  which support stm32 f411 microcontroller i tried ardupilot mission planner but its not working :( can you suggest mission planner software ?? or how to connect telemetry to inav ?

mursa1en, 2024-05-01:
> Added vl53l0x support. You will use I2C1 pins. You can test it. Write your result here.
> [inav_7.1.0_STM32F411CE.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/15138123/inav_7.1.0_STM32F411CE.zip)
> 

Vl53l1x successfully working sir 🫡[image]

Harshhhh16, 2024-05-01:
i connect telemetry but there is one issue that is i used inav 6.0 for fc in my case the battery voltage display 9.0v but i have use 3s 2200mah battery and also add 6.6k and 4.3k resistor to the port A1 is there is any additional setup in INAV ??

rizacelik, 2024-05-01:
INAV 7.1 support the bmi160.
Please see the "README" file.

mursa1en, 2024-05-03:
> Added vl53l0x support. You will use I2C1 pins. You can test it. Write your result here.
> [inav_7.1.0_STM32F411CE.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/15138123/inav_7.1.0_STM32F411CE.zip)
> 

Hello Sir 
The BMP280 barometer is not working over spi/i2c with this firmware but its working fine on others.

mursa1en, 2024-05-03:
> Added vl53l0x support. You will use I2C1 pins. You can test it. Write your result here.
> [inav_7.1.0_STM32F411CE.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/15138123/inav_7.1.0_STM32F411CE.zip)
> 

Hello Sir 
Sorry to bother you again. It seems like that the BMP 280 barometer can't be used in this firmware but it is working fine on the other firmwares [image]

[image]

[image]

mursa1en, 2024-05-03:
I connected it on the i2c pins. Btw i have connected a w25q64 so the nss1 and the the nss2 for the mpu6500 are occupied. Is the nss3 on the pin c14 usable?

rizacelik, 2024-05-03:
To use I2C, use the following CLI commands.

set baro_bustype = I2C
set baro_spi_device = 0
set baro_i2c_device = 1

rizacelik, 2024-05-03:
No, PC15 pin for SPI2 is BMP280 barometer NS pin. Use pin PC15 for barometer.

mursa1en, 2024-05-03:
Im getting these errors 
[image]

mursa1en, 2024-05-03:
Working fine over spi. Thanks for your help Sir

8305-kavin, 2025-01-09:
1.could you specify in which pins you have connected the bmp 280 and the sonar ?
2.Cant we use telemetry ?
3.Can you figure out a way to use 3dr telmetry insted of the black box?

rizacelik, 2025-01-09:
Please read the [README](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware#note-1) file. I gave links to telemetry above in this message. Read them too.