rizacelik/STM32F411CEU6_INAV_Firmware #54: Support for the BMI160 Gyro Acc

mursa1en, 2024-04-18:
Hello, Mr. Celik. Sorry to bother you again.
Do any of these firmware versions support the BMI160? If not, I kindly request that you consider incorporating support for the BMI160. The MPU6000 is no longer available, and the MPU6500 and MPU9250 suffer from increased noise levels. In contrast, the BMI160 overcomes these issues while maintaining a more cost-effective solution.

rizacelik, 2024-04-19:
I made the change. You can download the file.

mursa1en, 2024-04-19:
Thanks a lot sir

mursa1en, 2024-04-19:
[image]

This is how i connected the BMI 160. 

Inav is detecting the board as the MPU 6500 and not working. I tried both firmware. The 7.1 and the 7.1 hex

rizacelik, 2024-04-19:
Pins are correct. If you connected BMI160 but it detects it as MPU6500, this module is fake. INAV detects it automatically. There is no error in INAV.

[image]

closed 2024-04-21 by mursa1en