rizacelik/STM32F411CEU6_INAV_Firmware #55: Not recognizing USB for STM32F411CE

prathapjaga, 2024-05-04:
I used your [ target.h](https://github.com/prathapjaga/STM32F411CEU6_INAV_Firmware/blob/main/target.h) file as given your GitHub repository for compiling firmware. I ran into my local machine. I compiled the code successfully. I uploaded the firmware using DFU mode. as result, the USB is not recognized. but I used your [STM32F411CE_HEX.zip](https://github.com/prathapjaga/STM32F411CEU6_INAV_Firmware/blob/main/inav_6.1.1_STM32F411CE_HEX.zip), which worked well for the same board. could please help me with that? 
```
 81 | #define DEFAULT_RX_TYPE         RX_TYPE_PPM
      |                                 ^~~~~~~~~~~
/home/pratab-d/FCS/INAV_6.1.1/inav/src/main/config/parameter_group.h:229:9: note: in definition of macro 'PG_RESET_TEMPLATE'
  229 |         __VA_ARGS__                                                     \
      |         ^~~~~~~~~~~
/home/pratab-d/FCS/INAV_6.1.1/inav/src/main/rx/rx.c:113:21: note: in expansion of macro 'DEFAULT_RX_TYPE'
  113 |     .receiverType = DEFAULT_RX_TYPE,
      |                     ^~~~~~~~~~~~~~~
make[3]: *** [src/main/target/STM32F411CE/CMakeFiles/STM32F411CE.elf.dir/build.make:3778: src/main/target/STM32F411CE/CMakeFiles/STM32F411CE.elf.dir/__/__/rx/rx.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Makefile2:66384: src/main/target/STM32F411CE/CMakeFiles/STM32F411CE.elf.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:66417: src/main/target/STM32F411CE/CMakeFiles/STM32F411CE.dir/rule] Error 2
make: *** [Makefile:29712: STM32F411CE] Error 2
```
I got this issue solve by replace with `#define DEFAULT_RX_TYPE   RX_TYPE_SERIAL` in line no 81 after this I change get successful compile.

rizacelik, 2024-05-04:
https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware#attention-please

prathapjaga, 2024-05-04:
> https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware#attention-please

Thank you for your reply @rizacelik, I followed with all instruction as mentioned in the link. its work with STM32F411CE_HEX.hex. then I try to compile own by using target.h as given in same repository. but I did not work like your hex file  so please give me any suggestion on it

prathapjaga, 2024-05-04:
@rizacelik 
```
107 | #define DEFAULT_RX_TYPE         RX_TYPE_PPM
      |                                 ^~~~~~~~~~~
/home/pratab-d/FCS/INAV_7.1.0/inav/src/main/config/parameter_group.h:229:9: note: in definition of macro 'PG_RESET_TEMPLATE'
  229 |         __VA_ARGS__                                                     \
      |         ^~~~~~~~~~~
/home/pratab-d/FCS/INAV_7.1.0/inav/src/main/rx/rx.c:113:21: note: in expansion of macro 'DEFAULT_RX_TYPE'
  113 |     .receiverType = DEFAULT_RX_TYPE,

```
Now I am facing this type of issue, could you please tell me how I can solve it?

prathapjaga, 2024-05-04:
@rizacelik but firmware working with F411-Discovery board

rizacelik, 2024-05-04:
Change the lines as follows.

```
//#define USE_RX_PPM
#define DEFAULT_RX_TYPE RX_TYPE_SERIAL
```

Connect MPU9250 SPI2. It should work.

prathapjaga, 2024-05-06:
> Change the lines as follows.
> 
> ```
> //#define USE_RX_PPM
> #define DEFAULT_RX_TYPE RX_TYPE_SERIAL
> ```
> 
> Connect MPU9250 SPI2. It should work.

Thank you for your reply, I changed this line now is compile successfully , i  uploaded to the STM32F411CE but not work with it . actually work with STM32F411VE (DISCOVERY board ). please give any suggestion.

rizacelik, 2024-05-06:
I didn't do anything different. I am compiling the files. This is everything.
Replace the files prepared for the DISCOVERY board for yourself. You can look at the files I shared.

prathapjaga, 2024-05-07:
> I didn't do anything different. I am compiling the files. This is everything. Replace the files prepared for the DISCOVERY board for yourself. You can look at the files I shared.

Thank you for your kind reply, @rizacelik.  Thank you for spending time with me. I will try my own, and if I have any questions, may I contact you?

rizacelik, 2024-05-07:
Of course, I'll try to help.

prathapjaga, 2024-05-08:
Thank you , @rizacelik

prathapjaga, 2024-05-21:
hi @rizacelik, I am a fan of your work. I tried your firmware; it is flying well and flies very well. now like to build firmware like you. So kindly give me some valuable instructions for that. It may be helpful to enthusiastic people like me. Actually, I am working towards that goal. I completed some of the parts, but I am still not finished. It did not work out for me. I start my work from scratch. please help me.

prathapjaga, 2024-05-21:
> Of course, I'll try to help.

hi @rizacelik, I am a fan of your work. I tried your firmware; it is flying well and flies very well. now like to build firmware like you. So kindly give me some valuable instructions for that. It may be helpful to enthusiastic people like me. Actually, I am working towards that goal. I completed some of the parts, but I am still not finished. It did not work out for me. I start my work from scratch. please help me.