rizacelik/STM32F411CEU6_INAV_Firmware #56: request to upload your source files for BF-4.5 which is compiled by you.

vu2iia, 2024-05-08:
Hi Riza, 
I am using your BF-4.5.hex for my home-brew quad and its working fine (waiting for some parts to test). I have successfully compiled original BF-4.5 source and uploaded hex file, but USB VCP doesn't work. With your HEX file VCP & everything else works fine.  I want to compare your source file with original source, so that I can make required changes to suit my needs. My Rx is using PPM protocol, ESC is PWM type. Please guide.
Thank you,
Mahesh-vu2iia

vu2iia, 2024-05-08:
Hi Riza, 
It looks like I found the issue. Black pill F411 uses 25MHz crystal and config.h in [source code](https://github.com/betaflight/betaflight) is set to 8MHz, I changed the following line in config.h. Now my own compiled code is detected by BF-Configurator and VCP port is now working.
#define SYSTEM_HSE_MHZ 25    // 8--->25
Hope it helps someone like me.
Thanks,
mahesh-vu2iia

rizacelik, 2024-05-08:
I'm glad you solved the problem. Have a nice flight.