rizacelik/STM32F411CEU6_INAV_Firmware #57: PCB project

suchigno, 2024-05-15:
I designed a PCB for this great project, ordered with JLCPCB ($2 for 5 pieces, and $2 for shipping).
The PCB is only compatible for a quadcopter with wire for INAV 7.1.
The wiring diagram is in sections so it should be easy to understand.

Specs:
- micro: STM32F411CEU6 (obviously required)
- IMU: MPU9250/6500 (required, SPI connection)
- Barometer: BMP280 (optional, SPI connection)
- Compass: HMC5883 (optional, i2C connection)
- Buzzer: 5 volts continuous
- Receiving signal: PPM or SBUS (selected by jumper)
     SBUS: inverted and not (selected by jumper)
- Stip Led WS28xx

[image]
[image]

Some notes to help you understand:
NOTE: Always READ Rıza Çelik advice and suggestions.
NOTE: Always test everything first on a breadboard.
NOTE: Always remember to remove the propellers during setup tests.

NOTE Power Board
Power for the PCB components is taken from the ESC of Motor 1 (M1BR, back right).
Power for the LED strips is taken from the ESC of Motor 4 (M4FL, front left).
Never connect the USB cable if the M1BR ESC is connected to the board even if the battery is disconnected.
When you want to test the motors with the PCB connected to the configurator with the USB cable, use jumper cables to connect the M1BR ESC connector to the board, then connect only GND and Signal (therefore exclude the 5volts pin, the central one of the connector) .

NOTE Gyro Section
Even if you buy the MPU9250 module it probably won't be the original one, now out of production, so you will find the MPU6500 in it, so if you want a compass you have to add an HMC5883 module. Rıza Çelik reported this here [#42](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/issues/42).

NOTE Battery Monitor Section
In my case I power my drone with a 3S battery so the two resistors of the voltage divider (RP1 and RP2) are calculated to obtain approximately 3 volts from 12.6 volts. If you change battery cells, calculate the resistors to avoid damaging the board.

NOTE Receiver Section
My receiver only has PWM channels so I used a converter to have a SBUS. The pads for the PPM and SBUS connectors are used to connect it to the PCB.
In the images below you can see the converter I used and the pads for the connectors.
[image]
[image]

PPM
If your receiver has the PPM signal you can use the PPM pads, to power the receiver use the central pad which supplies the 5 volts. Then use the J-SBUS-PPM (pin 1 and 2) to connect pin A3 of the micro to the PPM signal.

SBUS
If your receiver has the SBUS signal you can use the SBUS pads, you have to take the 5 volts power supply from the PPM side to power it. You can also make a connection between the two central pads.
Then use the J-SBUS-PPM, pin 2 and 3 together.
If you do NOT need to invert the SBUS signal you can close the JSBUS which will bring the signal directly to pin A3 of the micro.
If you need to invert the SBUS signal you must leave the JSBUS disconnected and use the two resistors (RSBUS 4k7, RSBUS2 10K) and the BC547 transistor to invert the signal, as specified by Rıza Çelik.

NOTE Led Strip
In the PCB there are two connections for two LED strips, one is completely independent (Ext Strip Section), the other (strip section) is connected to pin A8 of the micro and can be used with the INAV configuration.
Remember that the LED strips are powered by the M4FL ESC.

Ext Strip Section
I needed to understand what the orientation of the drone was when it was overhead, so I added a strip under the frame that moves the LEDs from the last rear to the first front. The strip turns on/off with a switch and changes color with a tactile button.
The components used for this strip are:
- Attiny85 micro DIP
- SWNEO (switch on/off)
- CONNEO (connector for strip)
- SW (tactile button, I mounted a right angle)

If you don't want to use this feature just don't solder them to the PCB.
To program the attiny85 there are many guides on the web, you can create the LED effect you want.

Here you can download the zip file with the PCB gerbers
[Gerber_SMT32F411-FC-V2_PCB.zip](https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/files/15320611/Gerber_SMT32F411-FC-V2_PCB.zip)

rizacelik, 2024-05-15:
You really did a great job. Thank you for your contributions. This work will greatly simplify the use of the flight control card. Everything will be together and easy to use.
If you find it suitable, I will put your work in the warehouse as a PCB file at a convenient time on your behalf.

rizacelik, 2024-05-15:
It would be nice if you included HM-10 bluetooth connection with TX_PIN PA0 and RX_PIN PA1 via SOFTSERIAL2 in the scheme. Thus, configuration operations could be performed from the smartphone using the speedybee application.

suchigno, 2024-05-15:
Sure, I'm waiting to receive the JDY-08 module, when it arrives I'll test it and if it works, update the PCB.

suchigno, 2024-06-17:
Update PCB V2.1 with Software Serial Port 2 connector.
[Gerber_SMT32F411-FC-V2.1_2024-06-16.zip](https://github.com/user-attachments/files/15873538/Gerber_SMT32F411-FC-V2.1_2024-06-16.zip)

chynbnbnb, 2024-07-02:
good job!Thanks for your contribution

jpfree, 2024-08-14:
i user PCB V2.1, mpu9250-6500,but can't detects it....

dexterm, 2024-11-24:
@suchigno 
I printed your PCB, everything works fine, except that the receiver is not connecting, I am using flysky FS-i6X transmitter and receiver. It support PPM SBUS, pls guide me how to connect the pins for the receiver I read your notes on the receiver but still unable to get the receiver to work

Thanks
djcm95@gmail.com

suchigno, 2024-11-25:
Hi,
I have never used PPM, I used SBUS as indicated in the note.
I know the FS-i6X because I have one too, but what model of receiver are you using?
I can help you understand if the PCB works or not, but on the micro/INAV you should ask rizacelik.
To verify that there are no problems on the PCB you could check with a multimeter that the PPM (or SBUS) pad of the "J-SBUS-PPM" connector is correctly connected to pin A3 of the micro. If both arrive at the pin of the micro the problem is not the PCB.
Flysky uses the native iBUS protocol, then I think it can also set SBUS, but I have never tried. So make sure you can verify if the receiver sends SBUS protocol to the micro. To do this you could use an Arduino with an SBUS library that decodes the signals coming from the FS-i6X. There are many examples on Google for this.

dexterm, 2024-11-30:
> Hi, I have never used PPM, I used SBUS as indicated in the note. I know the FS-i6X because I have one too, but what model of receiver are you using? I can help you understand if the PCB works or [...]

Hi,
 Thank you for your quick response but I was late in responding, because I only devote time on my weekends. Anyway I verified with the multi-meter the JSBUS-PIN  is connected to pin A3 of the controller. Now let me test the receiver and share my feedback with you. BTW I am using FS-IA10B receiver

dexterm, 2024-12-08:
[image]
Hi,
I attached pics of the printed PCB, as mentioned I am using PPM ibus with flysky FSI6 receiver
FS-IA10B
I followed the video instructions of rizacelik but still unable to get transmitter to connect with receiver (it has binded with receiver successfully) 
If you observe one of the pics that I shared, I have soldered two resistors (RSBUS2) should I remove them ?
[image]

[image]

dexterm, 2024-12-08:
Hi
according to your documentation/instructions
..............
If you do NOT need to invert the SBUS signal you can close the JSBUS which will bring the signal directly to pin A3 of the micro.
If you need to invert the SBUS signal you must leave the JSBUS disconnected and use the two resistors (RSBUS 4k7, RSBUS2 10K) and the BC547 transistor to invert the signal, as specified by Rıza Çelik......

Since I am not using sbus should I remove the RS-BUS2 resistor and BCSBUS transistor ?

rizacelik, 2024-12-08:
If you are not using sbus, you do not need resistors and transistors. You can connect the radio receiver output directly to the A3 pin.

dexterm, 2024-12-15:
> If you are not using sbus, you do not need resistors and transistors. You can connect the radio receiver output directly to the A3 pin.

Thank you,
Finally got it to work with ppm-ibus using flysky receiver and transmitter

dexterm, 2024-12-21:
@suchigno
Any plans of integrating osd support in to your pcb