rizacelik/STM32F411CEU6_INAV_Firmware #58: Not getting com port

Farhat-s, 2024-05-19:
Hello sir. I have bought a stm32F411. But I am not able to upload any code into it. Cause I am not getting any com port for stm32 in my laptop. My laptop is based on windows 11. Can you please help me.

Farhat-s, 2024-05-22:
No sir

prathapjaga, 2024-05-22:
can you show your board ?

prathapjaga, 2024-05-22:
Then actually what shown in your device manager?

Farhat-s, 2024-05-23:
In device manager shows only serial bus devices ' stm32 bootloader'

prathapjaga, 2024-05-24:
actually, you're in the DFU (direct firmware upgrade) mode. your device works well for learn more use this below link
https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware

prathapjaga, 2024-05-24:
actually, always stm32 bootloader show in USB sector only, kindly recheck.

prathapjaga, 2024-05-24:
what IDE you have used ?

Farhat-s, 2024-05-24:
> actually, always stm32 bootloader show in USB sector only, kindly recheck.

yes.

Farhat-s, 2024-05-24:
> what IDE you have used ?

Arduino ide

prathapjaga, 2024-05-27:
If you are using the Arduino IDE for developing the code, stm32 has multiple ways to upload the code. After uploading the code by using DFU you need to mention in the serial setup must VCP (virtual comport) then only it will be recognized by the PC. my suggestion is going through   to some tutorial like " How to Use DFU mode in Arduino and how use VCP in stm32 by Arduino".