rizacelik/STM32F411CEU6_INAV_Firmware #62: Portable Configurator on phone and tablet

suchigno, 2024-06-17:
If you want to use the configurator on your phone or tablet you can do it using an app: SpeedyBee. (https://www.speedybee.com/speedy-bee-app/)

Available for Android and iOS, I think it is the best, complete and free to have an INAV compatible drone configurator.
There are 2 ways to connect the F411:

- using a USB OTG cable connected to your phone or tablet.
 Not very convenient because if you have to calibrate for example it has a cable that hinders your movements. If you connect it to the board you must remove the 5v power coming from the ESCs.

- using a BLE 4.0 module
I tested two different modules: HM-11 and HC-05.
[image]
[image]

NOTE: if you use the HM-11 module without adapter, small and light, you have to use a 5v to 3.3v converter, I used this one
[image]

In both cases you have to use AT commands to change the name of the module (optional), change the baurate to 57600.
You can use one of the board's serial ports, if you don't use GPS you can use the uart, or the software serial ports.
Here you can find the updated PCB where I added a connector for software serial 2.
The BLE modules consume little power, you can always leave it connected and use the app to, for example, check before the flight that the GPS has correctly configured the home point, way point, etc. without using the OSD.

rizacelik, 2024-06-17:
JDY-08 didn't work?

Thank you. I added your contributions to the readme file by naming them. Thank you very much for your valuable contributions.

suchigno, 2024-06-17:
No, unfortunately not. I changed the firmware to convert it to HM-10 protocol but the app finds it but doesn't connect.
However, there are two other cheap modules that work well.
Thanks

nopi890, 2024-07-16:
Awesome firmware, I tested with a JDY 31 serial sppc module, and it works 
Thank you very much