rizacelik/STM32F411CEU6_INAV_Firmware #63: 7.1 firmware W25Q128 SPI pin

AutoPlantBali, 2024-06-25:
hello @rizacelik,

for inav 7.1 wiring, I can use it like the old version inav wiring:
SPI1_NSS -> A4
SPI1_SCK_PIN -> A5
SPI1_MISO_PIN -> A6
SPI1_MOSI_PIN -> A7

solder the W25Q128 ic directly on the bottom side of the STM32F411CEU6 board
[image]

wssck1-cyber, 2025-08-28:
hi @rizacelik,
I used W25Q128 module which was working fine with INAV 7.1 firmware but since i soldered the W25Q128 ic on the bottom in similar way (as in above diagram) , Blackbox is not being detected. I even tried to write the firmware 7.1.2 by enabling two SPI's using target.c, target.h files which although being compiled successfully but still the STM32F411 doesn't respond at all. However once i switch back to 6.0.0 firmware, i could see Blackbox storage in INAV configurator, but since i am using PCB which is based on INAV 6.1.1 and higher version so couldn't continue with 6.0.0 firmware.
So, please guide me how to proceed further or if possible, kindly generate an .hex file based on 7.1.2 firmware with two SPIs configured i.e. SPI1 and SPI2 enabled.

rizacelik, 2025-08-28:
You can add built-in flash module support yourself. To do this, use the following codes.

```
#define USE_SPI
#define USE_SPI_DEVICE_1

#define SPI1_SCK_PIN            PA5
#define SPI1_MISO_PIN           PA6
#define SPI1_MOSI_PIN           PA7

#define USE_FLASHFS
#define USE_FLASH_M25P16
#define M25P16_CS_PIN           PA4
#define M25P16_SPI_BUS          BUS_SPI2
#define ENABLE_BLACKBOX_LOGGING_ON_SPIFLASH_BY_DEFAULT
```

The PA4, PA5,PA6,PA7 pins are used for the ADC. These pins should not be used for any other purpose. It would have been better if WeActStudio had assigned the onboard Flash module to pins SP2 instead of SP1. The board doesn't have many available pins.

wssck1-cyber, 2025-08-28:
@rizacelik, Thank you so much for your quick response.
I have modified the file as follows according to your instructions, I also comment out S5 from target.c as it was allocated to PA5, but board is still not responding even blue LED is also not blinking.

#pragma once

#define TARGET_BOARD_IDENTIFIER "ST41"

#define USBD_PRODUCT_STRING  "STM32F411CE"

#define LED0                    PC13
//#define LED1                    PC14

#define BEEPER                  PB1
#define BEEPER_INVERTED

// *************** SPI FLASH **************************

#define USE_SPI
#define USE_SPI_DEVICE_1

#define SPI1_SCK_PIN            PA5
#define SPI1_MISO_PIN           PA6
#define SPI1_MOSI_PIN           PA7

#define USE_FLASHFS
#define USE_FLASH_M25P16
#define M25P16_CS_PIN           PA4
#define M25P16_SPI_BUS          BUS_SPI1
#define ENABLE_BLACKBOX_LOGGING_ON_SPIFLASH_BY_DEFAULT

// *************** SPI1 Gyro & ACC **********************

// #define USE_SPI
#define USE_SPI_DEVICE_2

#define SPI2_SCK_PIN            PB13
#define SPI2_MISO_PIN           PB14
#define SPI2_MOSI_PIN           PB15

#define USE_IMU_MPU9250
#define MPU9250_SPI_BUS         BUS_SPI2
#define MPU9250_CS_PIN          PB12
#define IMU_MPU9250_ALIGN       CW0_DEG

#define USE_IMU_MPU6500
#define MPU6500_SPI_BUS         BUS_SPI2
#define MPU6500_CS_PIN          PB12
#define IMU_MPU6500_ALIGN       CW0_DEG

#define USE_IMU_BMI160
#define BMI160_SPI_BUS          BUS_SPI2
#define BMI160_CS_PIN           PB12
#define IMU_BMI160_ALIGN        CW0_DEG

#define USE_IMU_BMI270
#define BMI270_SPI_BUS          BUS_SPI2
#define BMI270_CS_PIN           PB12
#define IMU_BMI270_ALIGN        CW0_DEG

#define USE_IMU_ICM42605
#define ICM42605_SPI_BUS        BUS_SPI2
#define ICM42605_CS_PIN         PB12
#define IMU_ICM42605_ALIGN      CW0_DEG

#define USE_IMU_ICM20689
#define ICM20689_SPI_BUS        BUS_SPI2
#define ICM20689_CS_PIN         PB12
#define IMU_ICM20689_ALIGN      CW0_DEG

#define USE_BARO_BMP280
#define USE_BARO_SPI_BMP280
#define BMP280_SPI_BUS          BUS_SPI2
#define BMP280_CS_PIN           PC15

// *************** SPI2 OSD *****************************

#define USE_MAX7456
#define MAX7456_SPI_BUS         BUS_SPI2
#define MAX7456_CS_PIN          PC14

// *************** UART *****************************

#define USE_VCP

#define USE_UART1
#define UART1_TX_PIN            PA15
#define UART1_RX_PIN            PB3

#define USE_UART2
#define UART2_TX_PIN            PA2
#define UART2_RX_PIN            PA3

#define USE_SOFTSERIAL1
#define SOFTSERIAL_1_TX_PIN     PA9   // Softserial1_TX
#define SOFTSERIAL_1_RX_PIN     PA10  // Softserial1_RX

#if defined(STM32F411CE)

#define USE_SOFTSERIAL2
#define SOFTSERIAL_2_TX_PIN     PA0  // Softserial2_TX
#define SOFTSERIAL_2_RX_PIN     PA1  // Softserial2_TX
#define SERIAL_PORT_COUNT       5
#else
#define SERIAL_PORT_COUNT       4
#endif

#define USE_RX_PPM
#define DEFAULT_RX_TYPE         RX_TYPE_PPM
#define SERIALRX_PROVIDER       SERIALRX_SBUS
#define SERIALRX_UART           SERIAL_PORT_USART2

// *************** I2C /Baro/Mag/Pitot ********************
#define USE_I2C
#define USE_I2C_DEVICE_1
#define I2C1_SCL                PB8
#define I2C1_SDA                PB9

#define DEFAULT_I2C_BUS         BUS_I2C1

#define USE_IMU_MPU6050
#define IMU_MPU6050_ALIGN       CW0_DEG
#define MPU6050_I2C_BUS         BUS_I2C1

#define USE_BARO
#define BARO_I2C_BUS            BUS_I2C1
#define USE_BARO_ALL

#define USE_MAG
#define MAG_I2C_BUS             BUS_I2C1
#define USE_MAG_ALL

#define MAG_HMC5883_ALIGN CW0_DEG

// *************** ADC *****************************
#define USE_ADC
#define ADC_INSTANCE                ADC1
#define ADC1_DMA_STREAM             DMA2_Stream0
#define ADC_CHANNEL_1_PIN           PB0
//#define ADC_CHANNEL_2_PIN           PB1
//#define ADC_CHANNEL_2_PIN           PA6
//#define ADC_CHANNEL_3_PIN           PA7

#define VBAT_ADC_CHANNEL            ADC_CHN_1
//#define CURRENT_METER_ADC_CHANNEL   ADC_CHN_2
//#define RSSI_ADC_CHANNEL            ADC_CHN_2
//#define AIRSPEED_ADC_CHANNEL        ADC_CHN_3

#define USE_LED_STRIP
#define WS2811_PIN                  PA8

// ***************  OTHERS *************************

#define DEFAULT_FEATURES        (FEATURE_TX_PROF_SEL | FEATURE_VBAT | FEATURE_TELEMETRY | FEATURE_SOFTSERIAL)

#define USE_DSHOT
#define USE_ESC_SENSOR
#define USE_SERIAL_4WAY_BLHELI_INTERFACE

#define TARGET_IO_PORTA         0xffff
#define TARGET_IO_PORTB         0xffff
#define TARGET_IO_PORTC         0xffff
#define TARGET_IO_PORTD         (BIT(2))

#define MAX_PWM_OUTPUT_PORTS       11

rizacelik, 2025-08-28:
Extract the following codes and recompile them.

```
#define USE_SOFTSERIAL1
#define SOFTSERIAL_1_TX_PIN     PA9   // Softserial1_TX
#define SOFTSERIAL_1_RX_PIN     PA10  // Softserial1_RX

#if defined(STM32F411CE)

#define USE_SOFTSERIAL2
#define SOFTSERIAL_2_TX_PIN     PA0  // Softserial2_TX
#define SOFTSERIAL_2_RX_PIN     PA1  // Softserial2_TX
#define SERIAL_PORT_COUNT       5
#else
#define SERIAL_PORT_COUNT       4
#endif
```
SERIAL_PORT_COUNT value must be 3.

`#define SERIAL_PORT_COUNT       3`

wssck1-cyber, 2025-08-28:
Internal Flash working along with other sensors:

[image]