rizacelik/STM32F411CEU6_INAV_Firmware #64: Rebuild INAV

AutoPlantBali, 2024-06-26:
I tried to build inav last commit 7.1.2, using target.c and target.h in this repository, everything went successfully but when flashed to the black pill F411 board the blue led was not blinking and the COM port was not detected. are there any additional settings? I built it with the ninja command

vishal-gowda, 2024-07-04:
Hi @AutoPlantBali,
I had the same issue, It seems in my case one of the two blackpill did not work because it could be bad clone. But do note that the bad one in this case worked with STM32CubeIDE

thainuu, 2024-09-26:
> Tôi đã thử xây dựng inav lần cuối cùng là 7.1.2, sử dụng target.c và target.h trong kho lưu trữ này, mọi thứ đều diễn ra thành công nhưng khi flash vào bảng black pill F411, đèn LED màu xanh không nhấp nháy và cổng COM không được phát hiện. Có bất kỳ cài đặt bổ sung nào không? Tôi đã xây dựng nó bằng lệnh ninja

Have you managed to fix this issue yet? I am experiencing the same problem; could you help me?

kappyphm, 2025-10-30:
> I tried to build inav last commit 7.1.2, using target.c and target.h in this repository, everything went successfully but when flashed to the black pill F411 board the blue led was not blinking and the COM port was not detected. are there any additional settings? I built it with the ninja command

Same issue. My blue led worked but not blinking, my COM port is ok but it cannot connect to configurator. I connected my board with MPU6050.
@rizacelik could you help me :((((

rizacelik, 2025-10-30:
INAV 8.0 NO support PPM and MPU6050.

Which version did you use?
7.1 and older versions support the MPU6050. Check the pin connections. If you soldered the solder, the solder may not have been completely applied. You may have connected one of the pins incorrectly.