rizacelik/STM32F411CEU6_INAV_Firmware #68: Bmp 280 not working on iNav 7.1

ArshaanTech, 2024-08-04:
For some reason, my Barometer Bmp280 is not working on the updated version of iNav 7.1 
It was working fine on iNav 7.0 and betaflight 4.5

Here's a screenshot:

[image]

AaronSmith69, 2024-08-04:
what module did you use?

ArshaanTech, 2024-08-04:
The sensor? It is a BMP280 connected via i2c.

AaronSmith69, 2024-08-04:
Try using spi connection.

[image]

ArshaanTech, 2024-08-04:
I have it already connected on a pcb manner, I will have to change a lot of connections for that, I don't know why i2c isn't working on iNAV 7.1 It's working with previous version and betaflight.

ArshaanTech, 2024-08-05:
I tried it on spi connection too, no luck, same problem

bobiqritikol, 2025-02-17:
Maybe you have the bme280 not the bmp280. I made this mistake or buy new sensors some are faulty on purchasing