rizacelik/STM32F411CEU6_INAV_Firmware #74: Rangefinder

Lalbabu2001, 2024-12-08:
Hi, I my trying to connect rangefinder module VL53L0x module.
I soldered it to i2c  pins. But the configurator is still showing red.
I tested using Arduino and it's working fine but inav is not detecting it.
Please suggest fix.

rizacelik, 2024-12-08:
Test it and write the result here.
[inav_7.1.0_STM32F411CE.zip](https://github.com/user-attachments/files/18051568/inav_7.1.0_STM32F411CE.zip)

Lalbabu2001, 2024-12-08:
> Test it and write the result here. [inav_7.1.0_STM32F411CE.zip](https://github.com/user-attachments/files/18051568/inav_7.1.0_STM32F411CE.zip)

I tested this , The sensor is detected well but the readings are not showing. Attaching pic please verify.
[image]

rizacelik, 2024-12-08:
I don't have the VL53L0x sensor. I can't test it. There may be information on the internet or on YouTube about how to install it. If you can solve the problem, leave a note here. Maybe it will help someone.
The sensor you are using will give a value like 75 cm and it is very limited.

Lalbabu2001, 2024-12-09:
I want to have have a smooth landing for my quad , that's why I want to
install a rangefinder.
Another guy (eon claw)  who also made firmware for version 6.
I tested that it's working fine there.
But I don't trust him whether it will work reliably or not. It's less
matured than yours firmware.
I think there is some problem with the firmware,
Please check the i2c address and related stuffs from rangefinder in
firmware.

rizacelik, 2024-12-09:
Can you test this?
[inav_7.1.0_STM32F411CE.zip](https://github.com/user-attachments/files/18062741/inav_7.1.0_STM32F411CE.zip)

Lalbabu2001, 2024-12-09:
Oo I got it , when the gyro is  leveled (under 40°) only then it starts reading values & when gyro is not levelled I.e gyro is placed upside down then the rangefinder doesn't read any value.

I'm curious to know why this sensor is not working in version 6.1.1 and working fine in version 7.1?
Did you made some significant changes? What?

Thanks man for your support 👍🥰

Lalbabu2001, 2024-12-10:
The old 7.1.0 which is previous one also worked.
The last one you send also works,
I didn't find any difference in Hex size of both so I think both are same.
Did you made some changes in the last firmware?
The what changes you made ?