rizacelik/STM32F411CEU6_INAV_Firmware #75: BMP280 standalone cannot work on version 7.1

AaronSmith69, 2024-12-10:
GY-91 had no problem detecting the BMP280 but the sensor alone cannot be configurated on Inav 7.1. This issue has been raised for a long time ([https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware/issues/68](url)) and has not been fixed.

rizacelik, 2024-12-10:
I will also add INAV 8.0.

rizacelik, 2024-12-10:
There is BMP280 support. You need to connect BMP280 CS pin to C15 pin. I wrote this in the description section. I guess you didn't see it. You need to connect this sensor to SPI2 pins. Not I2C.
https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware#update-note-inav-710

AaronSmith69, 2024-12-15:
> There is BMP280 support. You need to connect BMP280 CS pin to C15 pin. I wrote this in the description section. I guess you didn't see it. You need to connect this sensor to SPI2 pins. Not I2C.
> https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware#update-note-inav-710

Sorry for taking a long time to reply. It was a malfunction fault of the sensor, the programme worked great after i switched to a new one. Thank you!

closed 2024-12-15 by AaronSmith69