rizacelik/STM32F411CEU6_INAV_Firmware #77: Do you support optical flowmeter

kloo4568, 2024-12-21:


kloo4568, 2024-12-21:
My optical flowmeter is UP-FLOW-LC-302-3C，The core chip of the module adopts the self-developed optical flow chip U30-5A.
Can you fit it on here？

kloo4568, 2024-12-21:
There's the msp protocol and 12c

rizacelik, 2024-12-21:
This sensor does not have INAV support.

kloo4568, 2024-12-21:
Thanks anyway

kloo4568, 2024-12-23:
> 此传感器不支持 INAV。

### Does the pwm3901 optical flow sensor be supported？

rizacelik, 2024-12-23:
INAV RANGEFINDER LISTS

https://github.com/iNavFlight/inav/blob/master/docs/Rangefinder.md

kloo4568, 2024-12-25:
> INAV 测距仪列表
> 
> https://github.com/iNavFlight/inav/blob/master/docs/Rangefinder.md

I use the officially supported vl53l0x rangefinder to connect to the same interface of the bmp280, but the sonar icon shows red, is this connection correct?
In addition, I have designed the pcb and integrated f411, mpu9250, bmp280, hmc5883 and w25q128, the size of which is only 6cm * 2cm.  I can share the optical flow module and laser ranging after integration, could you help me find the cheapest optical flow module and point out how to connect it? Thank you very much!
 A friends from China.

rizacelik, 2024-12-25:
Have you used INAV8.0? Use the latest version. Test and give feedback.
You can use vl53l0x rangefinder instead of optical flow module. INAV8.0 has this support.

If you want to use optical flow module, I need to make a separate compilation. Two separate modules do not work together. (for 3901 optical flow sensor)

kloo4568, 2024-12-25:
The inav8.0 I used connected vl5310x and bmp280 in parallel to the b8 and b9 interfaces, the sonar icon was still red, and the bmp280 also turned red. If the vl5310x was removed, the bmp280 was still red, and it was estimated that it was burned out

rizacelik, 2024-12-25:
Can you test this.

https://github.com/user-attachments/files/18062741/inav_7.1.0_STM32F411CE.zip