rizacelik/STM32F411CEU6_INAV_Firmware #78: Aircraft unlocked after a propeller does not turn problem

kloo4568, 2025-01-07:
Every time after unlocking, one of the four propellers will not turn, each time to fine-tune the pitch and roll rocker, until all four propellers turn to take off, I have turned off airmod, and set the self-stabilization.

rizacelik, 2025-01-07:
What version are you using and what exactly is the problem? I couldn't understand it from the message.

rizacelik, 2025-01-08:
Which motor is not working? 

Were 4 motor working in INAV 7.1? Did you try?

kloo4568, 2025-01-10:
4 motors always have one will not turn, need to slowly push the pitch rocker, it will turn, and airmod shows that it is off, but sometimes it will automatically open, slowly push the throttle rocker, the machine will directly turn over, this is inav8.0，inav7.0 has not been tested.